SpikeInterface/probeinterface #11: Handle shanks.

samuelgarcia, 2020-12-02:
@lfrank @bendichter @alejoe91 
Thank you for your feedbacks.

@lfrank @bendichter,  you both mention how to handle multi shank probe.

I already thought about it and it is not implemented yet because I have a doubt and want feedback.

Now there is 2 classes Probe and ProbeBunch.

Have in mind that the package have several goals:
 * describe probes : the more precesly but simple
 * describe wring to device when combinate several probes on devices (headstage wriing, missing channels).
  * make easy for for spike sorting algos to split a setup into several groups (here a shank tipically).

Way 1 : have more classes (ProbeBunch>Probe>Shank)
  * each shank would deal with it own geometry
  *  a Probe would an aggretor of Shank even for one shank probes

Way 2 : handle shank_ids as a vector inside Probe:
  * we could use an internal vector of shanks ids by probe to get channels and geometry for one shank.

There are pro and cons for each way.
I want to keep class model simple to have a simple and attractive doumentation.
My intuition would go for way 2 but I need feedback.

Thank you for helping.

bendichter, 2020-12-02:
I don't understand way 2. Could you elaborate?

alejoe91, 2020-12-02:
@bendichter instead of having a shank object, you would have a `shank_id` property for each electrode

bendichter, 2020-12-02:
@alejoe91 oh, I see. So that's fine as long as there are no shank attributes. If you want to define the geometry of a shank, position information, quality metrics, target brain region, or any other attributes, you'd need a shank object. I think it's likely we'll want shank attributes so I'd go with Way 1.

samuelgarcia, 2020-12-02:
I am not sure to anderstand you, Ben.

If we go for way1, the golden class will be "Shank" with its own geometry.
For me multi-shank Probe have a fixed and global geometry than include the multi shank.
For me we need a way to dscride vey easly one probe in one function.
If we go for 3 class hierachy I am afraid that making a probe object will be too verbose.

Maybe we can try a way in between where a probe handle internally shank_id and is able to create on the fly a Shank object than would have more or less all probe attributes for a sub channel set.
I can try to have Mixin class between Probe and Shank.
This could maybe lead to an over ingeneered solution (which is not a good idea for contributions, I think).

I also wanted to make the Probe object slicable maybe I will use the mechanism to generate sub shank.
I try something soon an dpropose a PR for this.

lfrank, 2020-12-02:
If it’s helpful, for our extension we have channel numbers that are within probe, so a probe with 4 shanks, 32 channels / shank has channel numbers 1-128, but to make it easier for the user to specify in a yaml file, they are grouped by shank (see example below). Given that one migth want to sort across shanks, I would be happy with any organization that maintains information about shanks.

Also, suggestion: change ProbeBunch to ProbeGroup

example yaml file (with … replacing most of the electrode records)

probe_type: 128c-4s8mm6cm-20um-40um-sl
units: 'um'
probe_description: '128 channel polyimide probe'
num_shanks: 4
contact_side_numbering: true
contact_size: 20.0
shanks:
  - shank_id: 0
    electrodes:
      - id: 0
        rel_x: 0
        rel_y: 0
        rel_z: 0
      - id: 1
        rel_x: 0
        rel_y: -40
        rel_z: 0
      - id: 2
        rel_x: 0
        rel_y: -80
        rel_z: 0
...
  - shank_id: 1
    electrodes:
      - id: 32
        rel_x: 344
        rel_y: 0
        rel_z: 0
      - id: 33
        rel_x: 344
        rel_y: -40
        rel_z: 0
      - id: 34
        rel_x: 344
        rel_y: -80
        rel_z: 0
… etc.

samuelgarcia, 2020-12-02:
@lfrank what you propose if the "way 2" in my previous proposal.
The electrode id is global to probe but they belong to diffrerent shank.

For me the how it is represented in file is something else.
We can handle several format in the probeinterface.io : readable (or no) shank centric or probe centric.

here my question is more the object hierachy.

samuelgarcia, 2020-12-04:
Shank started in #19
Need feedback.

samuelgarcia, 2020-12-04:
Here an example for handling shank:
https://probeinterface.readthedocs.io/en/main/examples/ex_04_multi_shank_probe.html#sphx-glr-examples-ex-04-multi-shank-probe-py

samuelgarcia, 2021-01-11:
I have relase today version 0.1.0 which handle Shank.

closed 2021-01-11 by samuelgarcia