SpikeInterface/probeinterface #12: Integration with NWB

samuelgarcia, 2020-12-02:
@lfrank @bendichter

You both mention NWB backend for handling probe geometry description.
I have NOT studied then yet.
I will.

Are theses two codes linked somehow ?

https://github.com/NovelaNeuro/ndx-franklab-novela
this seams to be a description of in house data management for recordings that also include probes description (with multi shank)

https://github.com/catalystneuro/ndx-spatial-coordinates/blob/main/src/spec/create_extension_spec.py
being unfamilir wuth nwb I get the purposed on this, could you explain more ?

Would be a way to define some kind of offcial to describe probes in NWB.
Maybe my question is naive because I am not aware of what have already been done.

samuelgarcia, 2020-12-02:
About the ndx-franklab-novela I will use lots of fields propose by this project.

bendichter, 2020-12-02:
ndx-spatial-coordinates is a work-in-progress. It's a project with the Allen Institute that encompasses several related topics: probe configuration, anatomical atlas, and transformations. We have started thinking about how to define objects in 2 and 3 d, so I thought it might be relevant for you. It's still a work-in-progress though, is likely to change, and currently has no documentation.

samuelgarcia, 2020-12-02:
Is it with Luke Campagnola ?

samuelgarcia, 2021-01-11:
Hi @bendichter.
I have release today version 0.1.0
The actual package contain a "read_nwb(...)" function which is empty at the moment.
When nwb format will have something about the spatial probe description it would be cool to implement this function as a proof of concept to test prointerface API.

CodyCBakerPhD, 2022-07-27:
@D1o0g9s will be helping with this task a bit 

Some pseudo-code design for reading NWB without the use of the extension @D1o0g9s is putting together, as well as a modular step-by-step bite-size tasklist

# 1

read stuff in probeinterface.io

```python
def read_from_nwb(nwbfile_path : str) -> Probe:  # or w/e name you want
with NWBHDF5IO(file, "r") as io:
     nwbfile = io.read()

    # figure out num_dims and num_contaacts,
    
    shape = nwbfile.electrodes.get("shape")
    # shape_params only necessary if shape is not None

    probe_object = Probe(ndim = num_dims, si_units="um")

    poisitions_2d = np.empty(shape=(num_contacts, num_dims))
    for j, column_name in enumerate(["rel_x", "rel_y", "rel_z"]):
           poisitions_2d[, j] = nwbfile.electrodes[column_name]

    probe_object.set_contacts(positioned_2d, shapes=shape, shape_params=shape_params)
    
    return probe_object
```

# 2

In neuroconv.tools.probeinterface

```python
def add_probe_info(nwbfile : NWBFile):
    # map all probeinterface.Probe fields to electrode columns + use the ndx-probe extension for nicely describing probe geometry in the new Device object

def write_probe(probe: Probe, metadata : dict, overwrite : bool = False):
    with NWBHDF5IO(file, "w" if overwrite else "r+") as io:
        nwbfile = make_nwbfile_from_metadata(metadata) if overwrite else io.read() # this is typical basic neuroconv metadata strcture, mainly for session_start_time
        add_probe_info(nwbfile=nwbfile)
        io.write()
```

# 3

read more complete information (if available) in probeinterface.io

```python
def read_from_nwb():  # or w/e name you want
with NWBHDF5IO(file, "r") as io:
     nwbfile = io.read()

     # all the previous stuff from #1

    # then read in all that stuff was that newly written through the use of # 2
    # goal being to be capable of a full roundtrip without any loss in information when using neuroconv.tools.probeinterface.write_probe to write a nwbfile, then using probe=read_from_nwb(nwbfile_path) to read it back
```

alejoe91, 2023-08-23:
https://github.com/SpikeInterface/ndx-probeinterface

lfrank, 2023-08-27:
> This Message Is From an External Sender
> This message came from outside your organization.
>
> https://github.com/SpikeInterface/ndx-probeinterface<https://urldefense.com/v3/__https://github.com/SpikeInterface/ndx-probeinterface__;!!LQC6Cpwp!vlc3DAXwtXJ8CJ2gEhvoTBjMzkFljZr8sN6ZMfdwJLn8CQHBN4_WnFU-XQjaFKqE6yzk7uk2efqxFFfkH-1z_vYzTt8$>

Great to see that this has come together…

CodyCBakerPhD, 2026-03-24:
@alejoe91 Asking for a separate reason now (BEP032 related)

Is there a way for ProbeInterface to 'read' a Probe object from an NWB file?