SpikeInterface/probeinterface #17: Serialization and formats

samuelgarcia, 2020-12-04:
I plan to write several format/serialisation:
  * binary based with h5 already started and working (will change a bit)
  * some human readable like json/ymal/msgpack based a on big dictionary

My idea would be to have the main collections (downlodable from somewhere) in binary h5 because it is smaller.
Then possible conversipon to other formats with io functions.

Any feedback ?

@rossant @jonnew

rossant, 2020-12-04:
Could you give an idea and post an example here of the contents of the file (field names, hierarchy, etc)?

samuelgarcia, 2020-12-04:
The hierachy will follow class hierachy. ProbeGroup>Probe
And attributes will same as the one in class.

electrode_positions
electrode_plane_axes
electrode_shapes
electrode_shape_params
device_channel_indices
electrode_ids
....

This is why I woul dfeedback on the naming/samantic itself for class name and attributes #13

samuelgarcia, 2020-12-04:
I will try to write write_yaml soon the example will be more easy.

rossant, 2020-12-04:
Do you need a new format, or would it be sufficient to (1) have a good in-memory data structure, (2) write import and export functions? If you have that you'll also have a universal conversion tool by the way.

samuelgarcia, 2020-12-04:
1. First  I want good a memmory structure representation in python. (Easy to use in other py package).
2. Then I want import/export to existing format (PRB, CSV ...)
3. Finally I want a way to serialize the entire memory structure to something quite universal (h5, json, ymal) to:
    * make everyone happy
    * build a downloadable collections of existing probes that contains all fields of the "memory structure" (= class attributes)

**3** is kind of a "new format" finally. No ?

@jonnew don't care about 1. but could be interest in 3 I think.

rossant, 2020-12-04:
> **3** is kind of a "new format" finally. No ?

Yes it is! If there has to be a new format, I would strongly favor a text-based format (json or yam) over a binary one (HDF5, npy, etc). Much easier to edit, human-readable, etc. And only if one of the existing formats cannot/should not be used as a standard. FWIW: I would not think PRB would make a good choice for a "standard" (we chose Python as a file format because it was the easiest at the time, but it was never meant to be a standard). Perhaps it would make sense to follow NWB nomenclature if a standard has to made.

samuelgarcia, 2020-12-04:
My main problem for human readable was if we have multi probe with sevral times 384 channels that themself handle with 6 to 8 vector (positions, shape, indices, shan_ids) then this "human readabl"e will be enormous and nobody will read it.
So finally it lead to enormous file that nobody reads.
Is is the case for PRB today, they are too big.

Maybe pure (hdf5) is also a good solution.

In any case I plan to implement **both** readable and binary here.
So anyone will be able to choose.
The main question is what will be the more "offcial" (for the collections).

rossant, 2020-12-04:
How big are these text files today?

jonnew, 2020-12-04:
What do you mean by enormous? The data these probes produce is on the order
of 100MB/second. Who cares how big the configuration file is?

All I want is some standardized, machine parsable format. Binary is not
good for this type of thing IMO because you want the file to be self
documenting. If JSON or XML is used, anybody will be able figure it out and
parse it.

jonnew, 2020-12-04:
So, I agree with Cyrille on all points.

samuelgarcia, 2020-12-04:
Enormous is several thouands lines. 
I known this is not big but several thousands is not "human readable".

OK for the self documenting.

So lets go JSON or XML or YAML for this format as default.
Any preference ?
I will do it very soon, so you can have a look.

rossant, 2020-12-04:
Since we probably don't need these files to be "human writable" I think we could go with JSON

samuelgarcia, 2020-12-07:
@rossant @jonnew 
Cyril and Jon, thankd a lot for your feedbacks.

I have switch to JSON here #20 
I leave this open for a while for otehrs feedback maybe.

I will also make some doc with a short example of the JSON generated file.
I will maybe add more keys for descriptions but main keys for the geometry are already here.
You could try it if you want.

samuelgarcia, 2020-12-07:
See also this https://probeinterface.readthedocs.io/en/main/examples/ex_06_import_export_to_file.html#sphx-glr-examples-ex-06-import-export-to-file-py

samuelgarcia, 2021-01-11:
@rossant @jonnew : I have relase today probeinterface 0.1.0.

After you feedback the format is now JSON based.
See https://probeinterface.readthedocs.io/en/main/format_spec.html

We have started a public probe library here : https://gin.g-node.org/spikeinterface/probeinterface_library

There only one probe at the moement there as a proof of concept.

We are in contact with some manufacturers to build the full library.

closed 2021-01-11 by samuelgarcia