SpikeInterface/probeinterface #39: "conctact" or "electrodes"

samuelgarcia, 2021-02-15:
@JuliaSprenger @alejoe91 @bendichter

In the discussion on BIDS extention [here](https://github.com/INCF/neuroscience-data-structure)
 julia suggested to replace "electrode" semantic by "contact".
Any thought about this ?

I think we should follow the same semantic here and in this BIDS extention.

samuelgarcia, 2021-02-15:
See this doc
https://docs.google.com/document/d/1oG-C8T-dWPqfVzL2W8HO3elWK8NIh2cOCPssRGv23n0/edit#

JuliaSprenger, 2021-02-15:
> See this doc
> https://docs.google.com/document/d/1oG-C8T-dWPqfVzL2W8HO3elWK8NIh2cOCPssRGv23n0/edit#

In the document we also started off with 'electrode', but in discussions with experimentalists 'contact' turned out to be better defined and less ambiguous than `electrode`, as an electrode can also be understood as having multiple 'contacts', e.g. tetrodes.

samuelgarcia, 2021-02-15:
Alessio, any comments ?

alejoe91, 2021-02-16:
Hi guys, both of them are fine in my opinion. If 'contact' received better feedback from experimentalists, we should go with that

samuelgarcia, 2021-02-24:
close with #44

closed 2021-02-24 by samuelgarcia