SpikeInterface/probeinterface #58: Neuropixel probe geometry

Shivooone, 2021-04-28:
Hello,

Is there a probe file for Neuropixel probes? I haven't found it on the library.

Thanks a lot,
Shiva

samuelgarcia, 2021-04-28:
neuropixel don't have fixed channel map but dynamic channel map since you can choose 384 channel mong 1000 physical contacts.
This is why it is not in the library.

But you have `probe = read_spikeglx('my_spike_glx_data.meta')` function that read the dynamic channel map from a spikeglx dataset.

If you recording come from open ephys, I don't know yet how the channel map channel positions is done but it should be also implemented.

closed 2021-05-06 by samuelgarcia

phanhuynh, 2022-02-14:
Hi Samuel: I know that there is a settings.xml file from the Open-Ephys recording folder but I don't know how to use it.  Do you have any advice for that?

samuelgarcia, 2022-02-14:
Could you open an issue here : https://github.com/SpikeInterface/probeinterface
and give use this file so we can implement the read oe probe directly ?