SpikeInterface/probeinterface #61: Setting channel groups on probe

timsainb, 2021-05-11:
How do you set the channels groups on a probe. e.g. if I'm using a neuronexus A4x2-tet-7mm-150-200-121, I have 4 shanks, with two groups of tetrodes per shank. How can I set channel_groups accordingly? I couldn't find it in the documentation.

Thanks!

samuelgarcia, 2021-05-11:
There is no grouping in probeinterface.
It is only for probe description.
The grouping is an "interpretation" for the sorting.

On spikeinterface side, you can do `Recording.set_probe(probe, group_mode='by_probe'/'by_shank')`.
On spikeinterface side you can even make grouping at whatever level you want, use it for slicing to get sub recording.
All probe information are also sliced and attached to subrecording.
Try it and make feedback on it it is a new feature.

timsainb, 2021-05-14:
Good to know, thanks!

closed 2021-05-14 by timsainb