SpikeInterface/probeinterface #69: Possible channel map discrepancy ASSY-77 H3 probe

rf13734, 2021-09-21:
Hi there!

samuelgarcia, 2021-09-22:
Hi @rf13734.
I think this is a known bug
See this https://gin.g-node.org/spikeinterface/probeinterface_library/issues/1

I think the problem is in the probe mapping json file and not the pathway (wiring.py).

I haven't yet merge the PR on gin because I would prefer to regenerate the entire probe collection (maybe some other probe ahev the same bug) with my own script.

I will try to do it soon.

before this could you download manually the json probe file in this PR : https://gin.g-node.org/spikeinterface/probeinterface_library/pulls/2

Then load it manuall with  `probe = read_probeinterface('my_probe_file.json')` instead of  no `probe = get_probe()`.
Then use the same path way and check that the PR fix your probem.

Also note that the best debegging is this:

```
plot_probe(probe,
                with_channel_index=False, with_contact_id=True, 
                with_device_index=True,
```

because **contact id** will be the "1-based" like cambrige neurotec and **device index** will be the "0-based" channel index in your intan headstage.

i hope iyt helps.

Sorry for this.

rf13734, 2021-09-22:
> Hi @rf13734.
> I think this is a known bug
> See this https://gin.g-node.org/spikeinterface/probeinterface_library/issues/1
> 
    From what I can tell, it would still not match the channel map from Cambridge Neurotech nor the 0-indexed version of this. Would that still be okay?
> I think the problem is in the probe mapping json file and not the pathway (wiring.py).
>
> I haven't yet merge the PR on gin because I would prefer to regenerate the entire probe collection (maybe some [...]

samuelgarcia, 2021-09-22:
#55 
@Enakcihc have help me to test and validate this. @Enakcihc : could you comment ?

The mapping which is tricky was also validated by @alejoe91 maybe we did it the wrong.

@rf13734 : could make the logic of pathway again blindly (without reading the doc you have and without our solution) just by following the pathway to check ?

Note that pathway logic do not depend on the probe but on connector only.
Here it is a 2 levels pathway and more error prone.

Could you also contact Thal from cambridge neurotec to check if he has a more recent version ?
I know that the doc in cambridge nerotec used to have an error also (and this is why we also have error here for the probe).

rf13734, 2021-09-23:
> #55
> @Enakcihc have help me to test and validate this. @Enakcihc : could you comment ?
> 
> The mapping which is tricky was also validated by @alejoe91 maybe we did it the wrong.
> 
> @rf13734 : could make the logic of pathway again blindly (without reading the doc you have and without our solution) just by following the pathway to check ?
>    As far as I understand it, one needs to assign the omnetics pin to the correct electrode contact point on the adaptor, so based on the schema adaptor-omnetics schema from CN, one manually assigns the electode to each inxx, e.g. in0 would receive from electrode 48 (If i remember correctly). From then, for kilosort at least, the channel mp depends on how one writes the bin file: for us we go from in0 to in63, so I think our channel map should be in the order of electrodes by ascending inxx, with the corresponding ycoords of these electrodes. Maybe I'm mistaken?

> Note that pathway logic do not depend on the probe but on connector only.
Is it the case that e.g. electrode 36 corresponds to contact 36 on the adaptor?

> Here it is a 2 levels pathway and more error prone.
> 
> Could you also contact Thal from cambridge neurotec to check if he has a more recent version ?
I have already contacted Tahl to see if he can give any pointers.

> I know that the doc in cambridge nerotec used to have an error also (and this is why we also have error here for the probe).
I noticed that  the error was fixed by Aaron Wong, but as far as I can tell, it would still result in a channel map that appears to have some 0-indexed channels, and some other 1-indexed channels?

Also, is the probe.device_channel_indices the inxx layout of the omnetics connector, or is it the connected channel map itself? (hope I'm making sense)

Thanks very much for looking into this, sorry for me being confused by many aspects of this pipeline!

Enakcihc, 2021-09-23:
Hi Ross, 
Indeed as @samuelgarcia mentioned, this is a two-step connection and is a bit complicated. Especially true is that there are two ways to plug in the omnetics connector between the Intan headstage and the adapter. Right now the connection "ASSY-77>Adpt.A64-0m32_2x-sm>RHD2164" matches the orientation we plugged in:  the headstage with the circuit board elements facing the probe side. In the end, we have a channel map that has the deepest electrode mapping to In26 (0-based).
bottom - 26, 24, 22, 19, 21, ... 42, 44, 41, 39, 37 - top
You may want to verify again with the plot_probe function, enabling both the contact id and device index
```
import spikeinterface
import probeinterface as pi
probeGroup = pi.read_probeinterface('ASSY-77-H3.json')
probe = probeGroup.probes[0]
probe.wiring_to_device('ASSY-77>Adpt.A64-Om32_2x-sm>RHD2164')
plotting.plot_probe(probe, with_contact_id=True, with_device_index=True)
```
The image should look something like this
[image]

`id` would be the contact point ID according to the CambridgeNeurotech specs. `dev` would be the index (0-based) on the Intan board.

Enakcihc, 2021-09-23:
Channel map given to kilosort2 is like this:
```
[chanMap',ycoords']

ans =

           1         940
           2         640
           3         920
           4         660
           5         900
           6         680
           7         880
           8         700
           9         860
          10         720
          11         840
          12         740
          13         820
          14         760
          15         800
          16        1160
          17        1060
          18        1140
          19        1080
          20        1120
          21        1260
          22         960
          23        1240
          24        1220
          25        1040
          26        1020
          27        1100
          28         980
          29        1180
          30        1200
          31         780
          32        1000
          33          40
          34         460
          35         240
          36         220
          37          20
          38         140
          39          60
          40          80
          41         260
          42         280
          43           0
          44         300
          45         160
          46         120
          47         180
          48         100
          49         200
          50         480
          51         440
          52         500
          53         420
          54         520
          55         400
          56         540
          57         380
          58         560
          59         360
          60         580
          61         340
          62         600
          63         320
          64         620

```

which seems to have the same values as yours, but in order.

samuelgarcia, 2021-09-24:
@rf13734
 * channel_index :  is only the order in which the conact are listed. It is not really important.
* contact_id : is the manufacturer "name" sometimes 1-based. Sometime this corespond to the omnetics by convinience but not 
  always
* device_channel_indices : is the channel at headstage (or more precisely recording device) level. It is always 0-based.

rf13734, 2021-09-24:
Hi @Enakcihc 

Hi Ross,
> Indeed as @samuelgarcia mentioned, this is a two-step connection and is a bit complicated. Especially true is that there are two ways to plug in the omnetics connector between the Intan headstage [...]

Ahh okay, I have in26 = electrode 60 rather than 43 (the deepest electrode on H3 probe). 

I have in46 = electrode 15 - I confirmed with Tahl that this is the preferred way to connect, but I guess it doesn't matter as long as the connections are tracked right?

[image]

> bottom - 26, 24, 22, 19, 21, ... 42, 44, 41, 39, 37 - top
> You may want to verify again with the plot_probe function, enabling both the contact id and device index
> 
> ```
> import spikeinterface
> import probeinterface as pi
> probeGroup = pi.read_probeinterface('ASSY-77-H3.json')
> probe = probeGroup.probes[0]
> probe.wiring_to_device('ASSY-77>Adpt.A64-Om32_2x-sm>RHD2164')
> plotting.plot_probe(probe, with_contact_id=True, with_device_index=True)
> ```
> 
> The image should look something like this
> [image]
> 
> `id` would be the contact point ID (1-based) according to the CambridgeNeurotech specs. `dev` would be the index (0-based) on the Intan board.

I see that there is a discrepancy with how we have assigned electrodes to intan pins, would I be right in thinking that for you in46 = electrode 31?

Based on the below schemas, I can only match in46 to electrode 15 or 38, and I can only match in26 to electrode 60 or 21, not 43. Have you maybe got a different headstage amplifier from the one below?
[image]
[image]

samuelgarcia, 2021-09-24:
I don't follow.
I think "in14" (intan) is plugged on "ometic 15"  no ?

rf13734, 2021-09-24:
> I don't follow.
> I think "in14" (intan) is plugged on "ometic 15" no ?

[image]
in14 to omnetic 15 Like this?

rf13734, 2021-09-24:
> Hi @samuelgarcia
> 
> > I don't follow.
> > I think "in14" (intan) is plugged on "ometic 15" no ?
> 
> [image]
> in14 to omnetic 15 Like this?

[image]

rf13734, 2021-09-24:
@samuelgarcia I have in14 paired with omnetic /electrode 38 instead.

Enakcihc, 2021-09-24:
Hi Ross,

We are using the same Intan headstage, but the channel map for the adaptor seems to be different from yours.

We have  the A64-OM32x2-sm Adaptor from NeuroNexus, with the following channel map.
[image]
[image]

Maybe you are using an adapter from a different manufacturer? 
I also thought of it being a combined channel map (adapter + probe) because the numbers on the samtec connector are identical to those of the H3 probe. However, mapping some of the channels out with our map also does not correspond to the numbers you have.

Assuming everything is correct, I would suggest it to be identified as a different kind of connection.