SpikeInterface/probeinterface #80: Problems loading the 'ASSY-116-H4' probe from cambridge neurotech

vincisLab, 2021-12-10:
Hi all,

we are experiencing some problems ,  importing the 'cambridgeneurotech','ASSY-116-H4' probe (we have no issue loading another probe we are using the ASSY-116-P-2).

What can be the issue?

Thanks!

--------------------------------------------------------------------------
This is the code:
import probeinterface as pi
from probeinterface import plotting
probe =pi.get_probe('cambridgeneurotech','ASSY-116-H4')
----------------------------------------------------------------------------
And this is Error code:

---------------------------------------------------------------------------
SSLCertVerificationError                  Traceback (most recent call last)
D:\AnacondaStuff\envs\si_env\lib\urllib\request.py in do_open(self, http_class, req, **http_conn_args)
   1353             try:
-> 1354                 h.request(req.get_method(), req.selector, req.data, headers,
   1355                           encode_chunked=req.has_header('Transfer-encoding'))

D:\AnacondaStuff\envs\si_env\lib\http\client.py in request(self, method, url, body, headers, encode_chunked)
   1255         """Send a complete request to the server."""
-> 1256         self._send_request(method, url, body, headers, encode_chunked)
   1257 

D:\AnacondaStuff\envs\si_env\lib\http\client.py in _send_request(self, method, url, body, headers, encode_chunked)
   1301             body = _encode(body, 'body')
-> 1302         self.endheaders(body, encode_chunked=encode_chunked)
   1303 

D:\AnacondaStuff\envs\si_env\lib\http\client.py in endheaders(self, message_body, encode_chunked)
   1250             raise CannotSendHeader()
-> 1251         self._send_output(message_body, encode_chunked=encode_chunked)
   1252 

D:\AnacondaStuff\envs\si_env\lib\http\client.py in _send_output(self, message_body, encode_chunked)
   1010         del self._buffer[:]
-> 1011         self.send(msg)
   1012 

D:\AnacondaStuff\envs\si_env\lib\http\client.py in send(self, data)
    950             if self.auto_open:
--> 951                 self.connect()
    952             else:

D:\AnacondaStuff\envs\si_env\lib\http\client.py in connect(self)
   1424 
-> 1425             self.sock = self._context.wrap_socket(self.sock,
   1426                                                   server_hostname=server_hostname)

D:\AnacondaStuff\envs\si_env\lib\ssl.py in wrap_socket(self, sock, server_side, do_handshake_on_connect, suppress_ragged_eofs, server_hostname, session)
    499         # ctx._wrap_socket()
--> 500         return self.sslsocket_class._create(
    501             sock=sock,

D:\AnacondaStuff\envs\si_env\lib\ssl.py in _create(cls, sock, server_side, do_handshake_on_connect, suppress_ragged_eofs, server_hostname, context, session)
   1039                         raise ValueError("do_handshake_on_connect should not be specified for non-blocking sockets")
-> 1040                     self.do_handshake()
   1041             except (OSError, ValueError):

D:\AnacondaStuff\envs\si_env\lib\ssl.py in do_handshake(self, block)
   1308                 self.settimeout(None)
-> 1309             self._sslobj.do_handshake()
   1310         finally:

SSLCertVerificationError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1131)

During handling of the above exception, another exception occurred:

URLError                                  Traceback (most recent call last)
C:\Users\VINCIS~1\AppData\Local\Temp/ipykernel_12436/1655949527.py in <module>
      1 #probe =pi.get_probe('cambridgeneurotech','ASSY-116-P-2')
----> 2 probe =pi.get_probe('cambridgeneurotech','ASSY-116-H4')
      3 print(probe)

D:\AnacondaStuff\envs\si_env\lib\site-packages\probeinterface\library.py in get_probe(manufacturer, probe_name)
     76 
     77     if probe is None:
---> 78         download_probeinterface_file(manufacturer, probe_name)
     79         probe = get_from_cache(manufacturer, probe_name)
     80 

D:\AnacondaStuff\envs\si_env\lib\site-packages\probeinterface\library.py in download_probeinterface_file(manufacturer, probe_name)
     32     localfile = cache_folder / manufacturer / (probe_name + '.json')
     33     distantfile = public_url + f'{manufacturer}/{probe_name}/{probe_name}.json'
---> 34     dist = urlopen(distantfile)
     35     with open(localfile, 'wb') as f:
     36         f.write(dist.read())

D:\AnacondaStuff\envs\si_env\lib\urllib\request.py in urlopen(url, data, timeout, cafile, capath, cadefault, context)
    220     else:
    221         opener = _opener
--> 222     return opener.open(url, data, timeout)
    223 
    224 def install_opener(opener):

D:\AnacondaStuff\envs\si_env\lib\urllib\request.py in open(self, fullurl, data, timeout)
    523 
    524         sys.audit('urllib.Request', req.full_url, req.data, req.headers, req.get_method())
--> 525         response = self._open(req, data)
    526 
    527         # post-process response

D:\AnacondaStuff\envs\si_env\lib\urllib\request.py in _open(self, req, data)
    540 
    541         protocol = req.type
--> 542         result = self._call_chain(self.handle_open, protocol, protocol +
    543                                   '_open', req)
    544         if result:

D:\AnacondaStuff\envs\si_env\lib\urllib\request.py in _call_chain(self, chain, kind, meth_name, *args)
    500         for handler in handlers:
    501             func = getattr(handler, meth_name)
--> 502             result = func(*args)
    503             if result is not None:
    504                 return result

D:\AnacondaStuff\envs\si_env\lib\urllib\request.py in https_open(self, req)
   1395 
   1396         def https_open(self, req):
-> 1397             return self.do_open(http.client.HTTPSConnection, req,
   1398                 context=self._context, check_hostname=self._check_hostname)
   1399 

D:\AnacondaStuff\envs\si_env\lib\urllib\request.py in do_open(self, http_class, req, **http_conn_args)
   1355                           encode_chunked=req.has_header('Transfer-encoding'))
   1356             except OSError as err: # timeout error
-> 1357                 raise URLError(err)
   1358             r = h.getresponse()
   1359         except:

URLError: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1131)>

samuelgarcia, 2021-12-13:
This sounds like a problem of certificate from https://gin.g-node.org/spikeinterface/probeinterface_library
Maybe the others probe were already downloaded in local cache.
Could you try again ?

vincisLab, 2021-12-15:
Hi Samuel, 

yes you are right. Now it is working.

closed 2021-12-15 by samuelgarcia