SpikeInterface/probeinterface #85: settings.xml file from Neuropixels, Open Ephys

phanhuynh, 2022-02-14:
Hello everyone, I'm using spikeinterface to analyze Neuropixel (1.0) data from Open Ephys.  However, I'm missing the ProbeInterface probe object.

Perhaps I'm supposed to use the settings.xml generated by OE but not sure how.  
[settings.txt](https://github.com/SpikeInterface/probeinterface/files/8062881/settings.txt) (I had to change the format from xml to txt in order to attach it)

I've also tried looking in ProbeInterface public library. And I'd like to avoid generating it from scratch. Does anyone know where I can find it?

Thank you! I'm new to this so your support is deeply appreciated.

samuelgarcia, 2022-02-15:
Hi @phanhuynh.
Thanks you for this file.
I don't see any obvious section in this file that would contain the conatct contact position on NP probe.
Could you dig a bit into the open-ephys doc and send us a link on how to retrieve this locations please ?

alejoe91, 2022-02-15:
I think that the channel locations are not stored in the settings for openephys. @jsiegle can you confirm this? In that case we can store all the locations for np1 and np2 probes in probeinterface and select the channels based on the settings

jsiegle, 2022-02-15:
As of version 0.3.3 of the Neuropixels PXI plugin, the channel locations (in microns from the tip) are stored in the `settings.xml` file. There's also a button within the the Neuropixels settings interface that directly exports the probe settings in ProbeInterface JSON format.

See [this page](https://github.com/open-ephys-plugins/neuropixels-pxi/releases) for full release notes.

phanhuynh, 2022-02-16:
@jsiegle, thank you!  I got the json file.
@alejoe91 , it's version 0.3.4-API7
 
Quick follow up:  what's the python code to set the probe in spikeinterface?  I tried:

probe = read_probeinterface('probe.json')
set_prob(probe)

Which doesn't work.  I feel like this should be obvious but I just can't find it online.  Thank you again.

alejoe91, 2022-02-16:
What's the full error?

phanhuynh, 2022-02-16:
---------------------------------------------------------------------------
NameError                                 Traceback (most recent call last)
~\AppData\Local\Temp/ipykernel_9584/4189760159.py in <module>
      1 probe = read_probeinterface('probe.json')
----> 2 set_prob(probe)

NameError: name 'set_prob' is not defined

phanhuynh, 2022-02-16:
Note, however, that read_probeinterface does work.  
[image]

alejoe91, 2022-02-16:
Ah! yeah there is a typo!

This is what you should do:
```
import spikeinterface.full as si
from probeinterface import read_probeinterface

recording = si.read_openephys("...")

probe = read_probeinterface('probe.json')
recording.set_probe(probe, in_place=True)

si.plot_probe_map(recording)
```

phanhuynh, 2022-02-16:
Unfortunately, it gives me this error:

---------------------------------------------------------------------------
AssertionError                            Traceback (most recent call last)
~\AppData\Local\Temp/ipykernel_9584/2040178160.py in <module>
      1 probe = read_probeinterface('probe.json')
----> 2 recording.set_probe(probe, in_place=True)

~\anaconda3\envs\si_env\lib\site-packages\spikeinterface\core\baserecording.py in set_probe(self, probe, group_mode, in_place)
    266         Wrapper on top on set_probes when there one unique probe.
    267         """
--> 268         assert isinstance(probe, Probe), 'must give Probe'
    269         probegroup = ProbeGroup()
    270         probegroup.add_probe(probe)

AssertionError: must give Probe

alejoe91, 2022-02-16:
Ah I see, it's a probe group. Use `recording.set_probegroup(...)` instead

phanhuynh, 2022-02-16:
I don't think I'm using a probe group (it's neuropixels), but here's the new error:
(note that there's a typo in the error message: set_proce)
---------------------------------------------------------------------------
Exception                                 Traceback (most recent call last)
~\AppData\Local\Temp/ipykernel_7760/3053197239.py in <module>
      1 probe = read_probeinterface('probe.json')
----> 2 recording.set_probegroup(probe, in_place=True)

~\anaconda3\envs\si_env\lib\site-packages\spikeinterface\core\baserecording.py in set_probegroup(self, probegroup, group_mode, in_place)
    272 
    273     def set_probegroup(self, probegroup, group_mode='by_probe', in_place=False):
--> 274         return self.set_probes(probegroup, group_mode=group_mode, in_place=in_place)
    275 
    276     def set_probes(self, probe_or_probegroup, group_mode='by_probe', in_place=False):

~\anaconda3\envs\si_env\lib\site-packages\spikeinterface\core\baserecording.py in set_probes(self, probe_or_probegroup, group_mode, in_place)
    344         if in_place:
    345             if not np.array_equal(new_channel_ids, self.get_channel_ids()):
--> 346                 raise Exception('set_proce(inplace=True) must have all channel indices')
    347             sub_recording = self
    348         else:

Exception: set_proce(inplace=True) must have all channel indices

alejoe91, 2022-02-16:
Can you share both the settings.xml and the probe.json?

phanhuynh, 2022-02-16:
Hi @alejoe91 , here is the json file (I had to convert to txt for github):
[probe.txt](https://github.com/SpikeInterface/probeinterface/files/8083929/probe.txt)

The settings.xml file can be found in my first post in this issue (also converted to txt).  
Thank you, again, so so much for all your attention :)

phanhuynh, 2022-02-18:
Hello @alejoe91 and @jsiegle , I just wanted to make sure that you're not waiting on me to provide anything.  Thank you for your help!

alejoe91, 2022-02-18:
@phanhuynh can you try this?

```
probegroup = read_probeinterface('probe.json')
probe = probegrouup.probes[0]
recording_w_prb = recording.set_probe(probe)
print(recording_w_prb) # this should have one channel less than 'recording'
```

or:
```
probegroup = read_probeinterface('probe.json')
recording_w_prb = recording.set_probegroup(probegroup)
print(recording_w_prb) # this should have one channel less than 'recording'
```

The problem is you cannot use `in_place` because the open-ephys NPIX recording has an extra channel which is the SY channel. The suggest approach instead should work because it selects a subset of channels.

phanhuynh, 2022-02-18:
Hi @alejoe91 , I tried both:
[image]
[image]

For both, when I apply the sorters (tridesclous and spykincircus), the error says "ValueError: There is not Probe attached to recording. use set_probe(...)" :

ValueError                                Traceback (most recent call last)
~\AppData\Local\Temp/ipykernel_11252/1173793720.py in <module>
      5 
      6 # parameters set by params dictionary
----> 7 sorting_TDC_5 = ss.run_tridesclous(recording=recording, output_folder='tmp_TDC_5',
      8                                    **default_TDC_params, )

~\anaconda3\envs\si_env\lib\site-packages\spikeinterface\sorters\runsorter.py in run_tridesclous(*args, **kwargs)
    410 
    411 def run_tridesclous(*args, **kwargs):
--> 412     return run_sorter('tridesclous', *args, **kwargs)
    413 
    414 

~\anaconda3\envs\si_env\lib\site-packages\spikeinterface\sorters\runsorter.py in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, with_output, **sorter_params)
     65                                     delete_output_folder=delete_output_folder, verbose=verbose,
     66                                     raise_error=raise_error, with_output=with_output, **sorter_params)
---> 67     return run_sorter_local(sorter_name, recording, output_folder=output_folder,
     68                             remove_existing_folder=remove_existing_folder,
     69                             delete_output_folder=delete_output_folder,

~\anaconda3\envs\si_env\lib\site-packages\spikeinterface\sorters\runsorter.py in run_sorter_local(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, **sorter_params)
     86     SorterClass.set_params_to_folder(
     87         recording, output_folder, sorter_params, verbose)
---> 88     SorterClass.setup_recording(recording, output_folder, verbose=verbose)
     89     SorterClass.run_from_folder(output_folder, raise_error, verbose)
     90     if with_output:

~\anaconda3\envs\si_env\lib\site-packages\spikeinterface\sorters\basesorter.py in setup_recording(cls, recording, output_folder, verbose)
    169             all_params = json.load(f)
    170             sorter_params = all_params['sorter_params']
--> 171         cls._setup_recording(recording, output_folder, sorter_params, verbose)
    172 
    173     @classmethod

~\anaconda3\envs\si_env\lib\site-packages\spikeinterface\sorters\tridesclous\tridesclous.py in _setup_recording(cls, recording, output_folder, params, verbose)
     86 
     87         # save prb file
---> 88         probegroup = recording.get_probegroup()
     89         prb_file = output_folder / 'probe.prb'
     90         write_prb(prb_file, probegroup)

~\anaconda3\envs\si_env\lib\site-packages\spikeinterface\core\baserecording.py in get_probegroup(self)
    398             positions = self.get_property('location')
    399             if positions is None:
--> 400                 raise ValueError('There is not Probe attached to recording. use set_probe(...)')
    401             else:
    402                 warn('There is no Probe attached to this recording. Creating a dummy one with contact positions')

ValueError: There is not Probe attached to recording. use set_probe(...)

samuelgarcia, 2022-02-18:
Hi,
 You have to run sorter with the `recording_w_prb` object.

phanhuynh, 2022-02-18:
Yes, thank you!  It seems to be working so far!  You guys are amazing!

closed 2022-02-18 by alejoe91