SpikeInterface/probeinterface #88: seeking json file for Buzsaki64 and/or Buzsaki32 probes

phanhuynh, 2022-03-16:
Hello SI community, has anyone created a probeinterface json file for NeuroNexus's Buzsaki64 or Buzsaki32 probes?  I would be deeply grateful!  (I'm using them through openephys but I don't think that's relevant)

samuelgarcia, 2022-03-17:
Hi Phan,
yes, at the moment neuronexus probe are not implemented in the library.
We get few contribution, if you make it please send us the json we can push it.
A simple way to do it is making an excel file with necessary columns and then use `Probe.from_dataframe()`

phanhuynh, 2022-03-23:
Hi @samuelgarcia , thank you for the tip.  I was able to do the following:

`buzsaki64probe = pd.read_csv('probeinterface/buzsaki64probe.csv')`
[buzsaki64probe.csv](https://github.com/SpikeInterface/probeinterface/files/8331878/buzsaki64probe.csv)
`buzsaki64probe = probeinterface.Probe.from_dataframe(buzsaki64probe)`
`plot_probe(buzsaki64probe)`
[image]
`recording = recording.set_probe(buzsaki64probe)`

The error occurs when I run the sorter:
`recording_saved = recording.save(folder="_preprocessed_NN", progress_bar=True, 
                                     n_jobs=4, total_memory="100M")`
`sorting_SPK = ss.run_sorter('spykingcircus', recording_saved, output_folder= '_output_SPK')`
[spikeinterface_log.txt](https://github.com/SpikeInterface/probeinterface/files/8331813/spikeinterface_log.txt)

The error message "No waveforms found! Are the data properly loaded?" makes me think that the issue is with the probe file, but I have no idea how to fix it.

Also, what's the difference between channel_index and contact_id?  And what's device_index?   
Thank you in advance!

samuelgarcia, 2022-03-23:
Hi Phan,
the probe looks good.
The difference between channel_index and contact_id is super impportant.
It related to the mapping between conact (number on the probe) and channel on your recording system (open ephys for instance.
Very often the "mapping" (aka wiring) is complex due to headstage internal wiring.
This mapping can be done manually or automatically if you use implemented headstage (intan one for instance).

You should carrefully read this:
  * https://probeinterface.readthedocs.io/en/main/examples/ex_05_device_channel_indices.html
  * https://probeinterface.readthedocs.io/en/main/examples/ex_11_automatic_wiring.html

The error message you have is related to spykingcircus. Maybe try an other sorter.
You should read this [tutorial](https://github.com/SpikeInterface/spiketutorials/tree/master/Official_Tutorial_SI_0.90_Aug2021) and see how you can make some plots to check that you have some spikes before launching a sorter.
Maybe you need some preprocessing before this.

phanhuynh, 2022-03-28:
Again, thank you @samuelgarcia!  I'm testing the probe file now.  Once I think it works, I'll upload it to the library (although, I'm not sure how many others will have the same configuration that I have.)

alejoe91, 2023-08-23:
@phanhuynh could we close this? Seems solver :)

phanhuynh, 2023-08-29:
Yes, thank you!

closed 2023-08-29 by alejoe91