SpikeInterface/probeinterface #92: Possible mismatch on cambridge neurotech ASSY-37-Fb

mfvd, 2022-03-18:
Hello,

I loaded the information of this probe and noticed that the channels ids do not match the ones found on cambridge neurotech's website (same map as the one provided with the probes).

Any idea what might be causing this mismatch?

[image]

[image]

alejoe91, 2022-03-18:
Hi @mfvd 

Thanks for the mismatch report. We'll double check with the CambridgeNeutotech team and let you know!

Cheers,
Alessio

mfvd, 2022-04-18:
Hi @alejoe91
Any news concerning this issue? I've also emailed cambridge neurotech but haven't heard from them yet.
Cheers

alejoe91, 2023-08-23:
@mfvd we recently received an updated version of the Cambridge catalog. Can you check if the issue is fixed now? The probeinterface library has been moved to GitHub here: https://github.com/SpikeInterface/probeinterface_library