SpikeInterface/probeinterface #95: Still problems loading probes from library

vincisLab, 2022-03-22:
Hello, 

we are still experiencing issues when trying to load probes from library.
For example

`import probeinterface as pi
from probeinterface import plotting

probe = pi.get_probe(manufacturer='neuronexus',
            probe_name='A1x32-Poly3-10mm-50-177')
`

will give 

---------------------------------------------------------------------------
SSLCertVerificationError                  Traceback (most recent call last)
File D:\AnacondaStuff\envs\full_spikeinterface_environment\lib\urllib\request.py:1354, in AbstractHTTPHandler.do_open(self, http_class, req, **http_conn_args)
   1353 try:
-> 1354     h.request(req.get_method(), req.selector, req.data, headers,
   1355               encode_chunked=req.has_header('Transfer-encoding'))
   1356 except OSError as err: # timeout error

File D:\AnacondaStuff\envs\full_spikeinterface_environment\lib\http\client.py:1256, in HTTPConnection.request(self, method, url, body, headers, encode_chunked)
   1255 """Send a complete request to the server."""
-> 1256 self._send_request(method, url, body, headers, encode_chunked)

File D:\AnacondaStuff\envs\full_spikeinterface_environment\lib\http\client.py:1302, in HTTPConnection._send_request(self, method, url, body, headers, encode_chunked)
   1301     body = _encode(body, 'body')
-> 1302 self.endheaders(body, encode_chunked=encode_chunked)

File D:\AnacondaStuff\envs\full_spikeinterface_environment\lib\http\client.py:1251, in HTTPConnection.endheaders(self, message_body, encode_chunked)
   1250     raise CannotSendHeader()
-> 1251 self._send_output(message_body, encode_chunked=encode_chunked)

File D:\AnacondaStuff\envs\full_spikeinterface_environment\lib\http\client.py:1011, in HTTPConnection._send_output(self, message_body, encode_chunked)
   1010 del self._buffer[:]
-> 1011 self.send(msg)
   1013 if message_body is not None:
   1014 
   1015     # create a consistent interface to message_body

File D:\AnacondaStuff\envs\full_spikeinterface_environment\lib\http\client.py:951, in HTTPConnection.send(self, data)
    950 if self.auto_open:
--> 951     self.connect()
    952 else:

File D:\AnacondaStuff\envs\full_spikeinterface_environment\lib\http\client.py:1425, in HTTPSConnection.connect(self)
   1423     server_hostname = self.host
-> 1425 self.sock = self._context.wrap_socket(self.sock,
   1426                                       server_hostname=server_hostname)

File D:\AnacondaStuff\envs\full_spikeinterface_environment\lib\ssl.py:500, in SSLContext.wrap_socket(self, sock, server_side, do_handshake_on_connect, suppress_ragged_eofs, server_hostname, session)
    494 def wrap_socket(self, sock, server_side=False,
    495                 do_handshake_on_connect=True,
    496                 suppress_ragged_eofs=True,
    497                 server_hostname=None, session=None):
    498     # SSLSocket class handles server_hostname encoding before it calls
    499     # ctx._wrap_socket()
--> 500     return self.sslsocket_class._create(
    501         sock=sock,
    502         server_side=server_side,
    503         do_handshake_on_connect=do_handshake_on_connect,
    504         suppress_ragged_eofs=suppress_ragged_eofs,
    505         server_hostname=server_hostname,
    506         context=self,
    507         session=session
    508     )

File D:\AnacondaStuff\envs\full_spikeinterface_environment\lib\ssl.py:1040, in SSLSocket._create(cls, sock, server_side, do_handshake_on_connect, suppress_ragged_eofs, server_hostname, context, session)
   1039             raise ValueError("do_handshake_on_connect should not be specified for non-blocking sockets")
-> 1040         self.do_handshake()
   1041 except (OSError, ValueError):

File D:\AnacondaStuff\envs\full_spikeinterface_environment\lib\ssl.py:1309, in SSLSocket.do_handshake(self, block)
   1308         self.settimeout(None)
-> 1309     self._sslobj.do_handshake()
   1310 finally:

SSLCertVerificationError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1131)

During handling of the above exception, another exception occurred:

URLError                                  Traceback (most recent call last)
Input In [44], in <module>
      2 from probeinterface import plotting
      3 #from probeinterface import Probe, get_probe
      4 #from probeinterface import get_probe
----> 5 probe = pi.get_probe(manufacturer='neuronexus',
      6             probe_name='A1x32-Poly3-10mm-50-177')

File D:\AnacondaStuff\envs\full_spikeinterface_environment\lib\site-packages\probeinterface\library.py:90, in get_probe(manufacturer, probe_name)
     87 probe = get_from_cache(manufacturer, probe_name)
     89 if probe is None:
---> 90     download_probeinterface_file(manufacturer, probe_name)
     91     probe = get_from_cache(manufacturer, probe_name)
     93 return probe

File D:\AnacondaStuff\envs\full_spikeinterface_environment\lib\site-packages\probeinterface\library.py:38, in download_probeinterface_file(manufacturer, probe_name)
     36 localfile = cache_folder / manufacturer / (probe_name + '.json')
     37 distantfile = public_url + f'{manufacturer}/{probe_name}/{probe_name}.json'
---> 38 dist = urlopen(distantfile)
     39 with open(localfile, 'wb') as f:
     40     f.write(dist.read())

File D:\AnacondaStuff\envs\full_spikeinterface_environment\lib\urllib\request.py:222, in urlopen(url, data, timeout, cafile, capath, cadefault, context)
    220 else:
    221     opener = _opener
--> 222 return opener.open(url, data, timeout)

File D:\AnacondaStuff\envs\full_spikeinterface_environment\lib\urllib\request.py:525, in OpenerDirector.open(self, fullurl, data, timeout)
    522     req = meth(req)
    524 sys.audit('urllib.Request', req.full_url, req.data, req.headers, req.get_method())
--> 525 response = self._open(req, data)
    527 # post-process response
    528 meth_name = protocol+"_response"

File D:\AnacondaStuff\envs\full_spikeinterface_environment\lib\urllib\request.py:542, in OpenerDirector._open(self, req, data)
    539     return result
    541 protocol = req.type
--> 542 result = self._call_chain(self.handle_open, protocol, protocol +
    543                           '_open', req)
    544 if result:
    545     return result

File D:\AnacondaStuff\envs\full_spikeinterface_environment\lib\urllib\request.py:502, in OpenerDirector._call_chain(self, chain, kind, meth_name, *args)
    500 for handler in handlers:
    501     func = getattr(handler, meth_name)
--> 502     result = func(*args)
    503     if result is not None:
    504         return result

File D:\AnacondaStuff\envs\full_spikeinterface_environment\lib\urllib\request.py:1397, in HTTPSHandler.https_open(self, req)
   1396 def https_open(self, req):
-> 1397     return self.do_open(http.client.HTTPSConnection, req,
   1398         context=self._context, check_hostname=self._check_hostname)

File D:\AnacondaStuff\envs\full_spikeinterface_environment\lib\urllib\request.py:1357, in AbstractHTTPHandler.do_open(self, http_class, req, **http_conn_args)
   1354         h.request(req.get_method(), req.selector, req.data, headers,
   1355                   encode_chunked=req.has_header('Transfer-encoding'))
   1356     except OSError as err: # timeout error
-> 1357         raise URLError(err)
   1358     r = h.getresponse()
   1359 except:

URLError: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1131)>
-----------------------------------------------------------------------------------------------------------------------------------

This happens also if we try to load Cabridge neurotech probes.

Thanks for the help!

alejoe91, 2022-08-05:
@vincisLab sorry to get back to this so late! Any update?? Were you able to doenload the probe from the library?

vincisLab, 2022-08-05:
Hi @alejoe91, yes now everything is working properly.

alejoe91, 2022-08-05:
Closing then!

closed 2022-08-05 by alejoe91

mzabolocki, 2023-08-18:
Hi all, I am having the same issue with macos. I am also trying to import Cambridge Neurotech probes (ASSY-77-P-2). 

Version for probe interface is: 0.2.17