SpikeInterface/probeinterface #100: Sorting doesn't take into consideration the loaded probe

benkiisma, 2022-03-30:
Hello,

In my code, I uploaded a predefined file containing information about my probe : 
```
channel_groups = {
    # Tetrode index
    0:
        {
        'channels': [15,0,16,31,17,30,2,12,18,28,14,29,1,19,5,9,21,25,13,3,27,20,11,24,7,4,23,22,10,26,6,8],

	'graph': [ 
(15,0),
(0,16),
(16,31),
(31,17),
(17,30),
(30,2),
(2,12),(2,18), (2,28),
(18,12),(18,28),
(12,28),
(28,14),
(14,29),
(29,1),
(1,19),
(19,5),
(5,21),(5,9),(5,25),
(21,9),(21,25),
(9,25),
(25,13),
(13,3),
(3,27),
(27,20),
(20,24),(20,11),(20,7),
(24,11),(24,7),
(11,7),
(7,4),
(4,23),
(23,22),
(22,26),(22,10),(22,6),
(26,10),(26,6),
(6,10),
(6,8)],

	'geometry': [[0,0],[0,-8],[0,-16],[0,-24],[0,-32],[0,-40],[0,-48],[2,-50],[-2,-50],[0,-52],[0,-60],[0,-68],[0,-76],[0,-84],[0,-92],[2,-94],[-2,-94],[0,-96],[0,-104],[0,-112],[0,-120],[0,-128],[2,-130],[-2,-130],[0,-132],[0,-140],[0,-148],[0,-156],[2,-158],[-2,-158],[0,-160],[0,-168]]
        }
}
```
After sorting, the resulting prb file is : 

```
total_nb_channels = 22
channel_groups = {
     0: 
        {
           'channels': [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21],
           'graph':  [(0, 1), (0, 2), (0, 3), (0, 4), (0, 5), (0, 6), (0, 7), (0, 8), (0, 9), (0, 10), (0, 11), (0, 12), (1, 2), (1, 3), (1, 4), (1, 5), (1, 6), (1, 7), (1, 8), (1, 9), (1, 10), (1, 11), (1, 12), (1, 13), (2, 3), (2, 4), (2, 5), (2, 6), (2, 7), (2, 8), (2, 9), (2, 10), (2, 11), (2, 12), (2, 13), (2, 14), (3, 4), (3, 5), (3, 6), (3, 7), (3, 8), (3, 9), (3, 10), (3, 11), (3, 12), (3, 13), (3, 14), (3, 15), (4, 5), (4, 6), (4, 7), (4, 8), (4, 9), (4, 10), (4, 11), (4, 12), (4, 13), (4, 14), (4, 15), (4, 16), (5, 6), (5, 7), (5, 8), (5, 9), (5, 10), (5, 11), (5, 12), (5, 13), (5, 14), (5, 15), (5, 16), (5, 17), (6, 7), (6, 8), (6, 9), (6, 10), (6, 11), (6, 12), (6, 13), (6, 14), (6, 15), (6, 16), (6, 17), (6, 18), (7, 8), (7, 9), (7, 10), (7, 11), (7, 12), (7, 13), (7, 14), (7, 15), (7, 16), (7, 17), (7, 18), (8, 9), (8, 10), (8, 11), (8, 12), (8, 13), (8, 14), (8, 15), (8, 16), (8, 17), (8, 18), (8, 19), (9, 10), (9, 11), (9, 12), (9, 13), (9, 14), (9, 15), (9, 16), (9, 17), (9, 18), (9, 19), (9, 20), (9, 21), (10, 11), (10, 12), (10, 13), (10, 14), (10, 15), (10, 16), (10, 17), (10, 18), (10, 19), (10, 20), (10, 21), (11, 12), (11, 13), (11, 14), (11, 15), (11, 16), (11, 17), (11, 18), (11, 19), (11, 20), (11, 21), (12, 13), (12, 14), (12, 15), (12, 16), (12, 17), (12, 18), (12, 19), (12, 20), (12, 21), (13, 14), (13, 15), (13, 16), (13, 17), (13, 18), (13, 19), (13, 20), (13, 21), (14, 15), (14, 16), (14, 17), (14, 18), (14, 19), (14, 20), (14, 21), (15, 16), (15, 17), (15, 18), (15, 19), (15, 20), (15, 21), (16, 17), (16, 18), (16, 19), (16, 20), (16, 21), (17, 18), (17, 19), (17, 20), (17, 21), (18, 19), (18, 20), (18, 21), (19, 20), (19, 21), (20, 21)],
           'geometry':  {
               0: [0.0, 0.0],
               1: [0.0, -8.0],
               2: [0.0, -24.0],
               3: [0.0, -32.0],
               4: [0.0, -40.0],
               5: [0.0, -48.0],
               6: [2.0, -50.0],
               7: [-2.0, -50.0],
               8: [0.0, -60.0],
               9: [0.0, -84.0],
               10: [0.0, -92.0],
               11: [2.0, -94.0],
               12: [0.0, -96.0],
               13: [0.0, -104.0],
               14: [0.0, -112.0],
               15: [2.0, -130.0],
               16: [0.0, -132.0],
               17: [0.0, -140.0],
               18: [0.0, -148.0],
               19: [2.0, -158.0],
               20: [0.0, -160.0],
               21: [0.0, -168.0],
           }
       },
}
```

This leads to a wrong analysis of the data ! Do you know from what the issue could come?

Thank you very much for your help.

alejoe91, 2022-03-30:
Hi, can you share the entire code so we can take a look? Internally, when you load a probe object the probe is sorted in a linear way, but positions should be as well. So it would be nice to take a look at the entire example!

alejoe91, 2022-03-30:
You need to pass `recording_prb` to the `run_klusta()` function, not `recording`!

samuelgarcia, 2022-03-30:
`load_probe_file` is the old spikeinetrface API. The old API do not use probeinterface at all.
You should go for the new one.

benkiisma, 2022-03-30:
> You need to pass `recording_prb` to the `run_klusta()` function, not `recording`!

You mean I have to add recording_probe to the parameters of 'run_klusta()' ?? Because if I replace it, how can klusta sort the data if I don't give the data to the function ?  'recording' is the variable containing the recorded data while 'recording_prb' is just the geometry and the graphs of the probe !

benkiisma, 2022-03-30:
> `load_probe_file` is the old spikeinetrface API. The old API do not use probeinterface at all.
> You should go for the new one.

Yes !! I am trying to switch from the old API to the new one !!

benkiisma, 2022-03-30:
> You can check a tutorial here: https://spikeinterface.readthedocs.io/en/latest/getting_started/plot_getting_started.html
> 
> And you can load a `prb` file with the `probeinterface.read_prb()` function!

Thank you, I will try this !!

benkiisma, 2022-04-02:
Hello, I tried to apply the function you proposed and read the tutorial but I still have an error when trying to link the probe to my recording. 
Here is the code I use : 
```
probe = read_prb('probe.prb')
print(probe)
probe.to_dataframe(complete=True).loc[:, ["contact_ids" ,"device_channel_indices"]]
plot_probe_group(probe)
recording_prb = recording.set_probe(probe)
```
**The error is :** 
```
---------------------------------------------------------------------------
AssertionError                            Traceback (most recent call last)
Input In [37], in <cell line: 1>()
----> 1 recording_prb = recording.set_probe(probe)

File ~/opt/anaconda3/envs/si090/lib/python3.8/site-packages/spikeinterface/core/baserecording.py:268, in BaseRecording.set_probe(self, probe, group_mode, in_place)
    264 def set_probe(self, probe, group_mode='by_probe', in_place=False):
    265     """
    266     Wrapper on top on set_probes when there one unique probe.
    267     """
--> 268     assert isinstance(probe, Probe), 'must give Probe'
    269     probegroup = ProbeGroup()
    270     probegroup.add_probe(probe)

AssertionError: must give Probe
```

alejoe91, 2022-04-02:
So the `read_prb()` function returns a `ProbeGroup` because it can describe multiple probes. You can use the `recording.set_probegroup(probe)` instead!

benkiisma, 2022-04-02:
Thank you @alejoe91 I can set de recording probe now !! However, after sorting, the probe is still not taken into consideration. Here is my probe before sorting: 
```
channel_groups = {
    # Tetrode index
    0:
        {
        'channels': [15,0,16,31,17,30,2,12,18,28,14,29,1,19,5,9,21,25,13,3,27,20,11,24,7,4,23,22,10,26,6,8],

	'graph': [ 
(15,0),
(0,16),
(16,31),
(31,17),
(17,30),
(30,2),
(2,12),(2,18), (2,28),
(18,12),(18,28),
(12,28),
(28,14),
(14,29),
(29,1),
(1,19),
(19,5),
(5,21),(5,9),(5,25),
(21,9),(21,25),
(9,25),
(25,13),
(13,3),
(3,27),
(27,20),
(20,24),(20,11),(20,7),
(24,11),(24,7),
(11,7),
(7,4),
(4,23),
(23,22),
(22,26),(22,10),(22,6),
(26,10),(26,6),
(6,10),
(6,8)],

	'geometry': [[0,0],[0,-8],[0,-16],[0,-24],[0,-32],[0,-40],[0,-48],[2,-50],[-2,-50],[0,-52],[0,-60],[0,-68],[0,-76],[0,-84],[0,-92],[2,-94],[-2,-94],[0,-96],[0,-104],[0,-112],[0,-120],[0,-128],[2,-130],[-2,-130],[0,-132],[0,-140],[0,-148],[0,-156],[2,-158],[-2,-158],[0,-160],[0,-168]]
        }
}
```

And here it is after sorting : 

```
channel_groups = {
    0:
        {
           'channels': [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31],
           'geometry':  {
               0: [0.0, 0.0],
               1: [0.0, -8.0],
               2: [0.0, -16.0],
               3: [0.0, -24.0],
               4: [0.0, -32.0],
               5: [0.0, -40.0],
               6: [0.0, -48.0],
               7: [2.0, -50.0],
               8: [-2.0, -50.0],
               9: [0.0, -52.0],
               10: [0.0, -60.0],
               11: [0.0, -68.0],
               12: [0.0, -76.0],
               13: [0.0, -84.0],
               14: [0.0, -92.0],
               15: [2.0, -94.0],
               16: [-2.0, -94.0],
               17: [0.0, -96.0],
               18: [0.0, -104.0],
               19: [0.0, -112.0],
               20: [0.0, -120.0],
               21: [0.0, -128.0],
               22: [2.0, -130.0],
               23: [-2.0, -130.0],
               24: [0.0, -132.0],
               25: [0.0, -140.0],
               26: [0.0, -148.0],
               27: [0.0, -156.0],
               28: [2.0, -158.0],
               29: [-2.0, -158.0],
               30: [0.0, -160.0],
               31: [0.0, -168.0],
           }
       },
}
```

As you can see, the 'graph' part is not taken into consideration.

alejoe91, 2022-04-02:
The graph part is only used by klusta. Once loaded to spikeinterface, the other sorters directly use the loaded information with the channel locations.

benkiisma, 2022-04-02:
I understand. So as klusta is not running on spikeinterface anymore, I will have to run it using the docker to get the graph part into consideration, right ?

alejoe91, 2022-04-02:
You could install python 3.7 and install klusta in it. But with later Python version only the docker solution is available

closed 2022-08-05 by alejoe91