SpikeInterface/probeinterface #102: In WaveformView, channel locations incorrect

phanhuynh, 2022-04-06:
Hello ! First, thank you for your help! 
In WaveformView, the location of the channels do not match the location I entered into the probeinterface.
Here is the probe file I put into spikeinterface:
[buzsaki64probe.csv](https://github.com/cortex-lab/phy/files/8428890/buzsaki64probe.csv)

[image]

[image]

phanhuynh, 2022-04-12:
Hello folks, any ideas?

phanhuynh, 2022-04-12:
Thank you so much, @alejoe91 
[image]
[nnexus.txt](https://github.com/SpikeInterface/probeinterface/files/8474506/nnexus.txt)

alejoe91, 2022-04-12:
Can you check that the `recording_saved` has the correct probe info by plotting the probe map for it?

alejoe91, 2022-04-12:
Also note that you are not loading the separate shanks as groups. to use that you have to use the `set_probe(probe, group_mode="by_shank")` option. (Not sure if this is intentional)

phanhuynh, 2022-04-13:
The recording_saved looks good:
[image]

However, I could not execute set_probe(probe, group_mode="by_shank")
---------------------------------------------------------------------------
```
AttributeError                            Traceback (most recent call last)
Input In [74], in <module>
----> 1 probeinterface.set_probe(buzsaki64probe, group_mode="by_shank")

AttributeError: module 'probeinterface' has no attribute 'set_probe'
```

alejoe91, 2022-04-13:
You should use `recording.set_probe(buzsaki64probe, group_mode="by_shank", in_place=True)`

phanhuynh, 2022-04-13:
Thank you!  There are no errors this time, but the original problem persists:
[image]

alejoe91, 2022-04-13:
Can you load the `channel_positions.npy` in the Phy folder and plot them?

phanhuynh, 2022-04-13:
It seems to work:
[image]

alejoe91, 2022-04-13:
Can you try to add `plt.axis("equal")`?

phanhuynh, 2022-04-13:
[image]

alejoe91, 2022-04-13:
so positions are correct! It's just a scaling issue ;)

alejoe91, 2022-04-13:
Did you load with `group_mode="by_shank"`?

phanhuynh, 2022-04-13:
Yes

alejoe91, 2022-04-13:
you should have a column called `group`. The `sh` column is generated internally by Phy and we don't have control over it

phanhuynh, 2022-04-13:
I added a column called group, but I don't see any difference.  Is it possible that it's not necessary?

phanhuynh, 2022-04-13:
I see, thank you!  Do you have advice on how to set the groupings correctly:
[image]

alejoe91, 2022-04-13:
that's weird. Are you sure you resaved the recording after loading the probe with `by_shank`? You can check: `recording_saved.get_channel_groups()`

phanhuynh, 2022-04-14:
It's weird, indeed.  I've deleted the previous folders and re-ran the whole script:
[image]
And yet it still doesn't group:
[image]

alejoe91, 2022-04-14:
Ok then this might be a deeper bug, maybe related to the order in which we load probe and properties. I'll take a closer look :)

samuelgarcia, 2022-04-14:
Maybe it is the set_probe(inplace=True).
inplace=True is not the reommended way.
`rec = rec.set_probe(inplace=False)` is better but maybe this is unrelated.

alejoe91, 2022-04-14:
No I think this is related to the fact that the probe is loaded as extra_metadata and we don't propagate the `group_mode`...

alejoe91, 2022-04-14:
@phanhuynh I checked on my side and the channel groups are propagated as supposed to be:
[image]

Maybe try to remove the folder with the recording or use a different name. It should be working.

phanhuynh, 2022-04-14:
Perhaps your test is different because you recording is already grouped.
Mine is not grouped:
[image]

Furthermore, there is still no grouping after I set_probe:
[image]

alejoe91, 2022-04-14:
You are not using `group_mode='by_shank'`! This is how I added groups

alejoe91, 2022-04-14:
Or actually you are using set_probe twice. The first one is enough

phanhuynh, 2022-04-15:
Good catch!  Thank you.  So now the grouping is working: 
[image]

But not in Phy:
[image]

However, it's not clear to me that it matters.  Does it?

phanhuynh, 2022-04-19:
Hello again, sorry to take up so much of your time.  Just curious whether it matters if sh and channel_group remains all 0s.  Thank you!

alejoe91, 2022-04-19:
no, that's internal to Phy ;) channel groups are loaded back to spikeinterface with the `read_phy` function, so the `sh` column is really irrelevant! 

Let me know if I can close the issue then!

phanhuynh, 2022-04-19:
wonderful.  Thank you so much!  Yes, feel free to close :)

closed 2022-04-20 by alejoe91