SpikeInterface/probeinterface #106: Load Open-ephys probe data with multi-shank

MarineChap, 2022-05-12:
Hello,
I am trying to load a multishank recording like this but it does not seems to work. 

Based on this comment in read_openephys,    _# TODO get example of multishank for parsing_ ([L911](https://github.com/SpikeInterface/probeinterface/blob/main/probeinterface/io.py#L911)), does that means we cannot load yet multishank probe from open-ephys ? Or the problem is elsewhere? 

```
rec_oe1 = si.read_openephys(oe_npx_folder)
rec_oe1.get_probe()

---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
/tmp/ipykernel_161120/3307797951.py in <module>
      1 # internal list of segments- don't touch this!
----> 2 rec_oe1.get_probe()

~/miniconda3/envs/fkdev/lib/python3.7/site-packages/spikeinterface/core/baserecording.py in get_probe(self)
    449 
    450     def get_probe(self):
--> 451         probes = self.get_probes()
    452         assert len(probes) == 1, 'there are several probe use .get_probes() or get_probegroup()'
    453         return probes[0]

~/miniconda3/envs/fkdev/lib/python3.7/site-packages/spikeinterface/core/baserecording.py in get_probes(self)
    454 
    455     def get_probes(self):
--> 456         probegroup = self.get_probegroup()
    457         return probegroup.probes
    458 

~/miniconda3/envs/fkdev/lib/python3.7/site-packages/spikeinterface/core/baserecording.py in get_probegroup(self)
    462             positions = self.get_property('location')
    463             if positions is None:
--> 464                 raise ValueError('There is not Probe attached to recording. use set_probe(...)')
    465             else:
    466                 warn('There is no Probe attached to this recording. Creating a dummy one with contact positions')

ValueError: There is not Probe attached to recording. use set_probe(...)

```

Probe interface version : 0.2.9
[settings.txt](https://github.com/SpikeInterface/probeinterface/files/8679066/settings.txt)

Thanks, Have a good day
Marine, Fklab/Nerf

alejoe91, 2022-05-12:
Hi Marine, thanks for the report. I can't see any shank info in the settings file..however, the probe shoulf be loaded automatically. Can you try to run this?

```
import probeinterface as pi

probe_oe = pi.read_open_ephys("path-to-oe-folder")
```

and let me know if you get errors?

alejoe91, 2022-05-12:
Thanks for checking! That's weird because you are indeed using the Neuropic-PXI plugin! 

The SpikeInterface only tries to load the probe, but if it fails it doesn't throw an error. At least we should display a warning though!

I'll try to reproduce and fix the error from the settings file you provided :)

MarineChap, 2022-05-12:
The detection of the processor is based on the name which for me is different (Sources/Neuropix-PXI) instead of Neuropix-PXI. I will investigate why mine is different. I think it is due to how I installed it (develop mode) but anyway I think you would be better checking on the plugingName attribute which will stay constant. 
After the hackathon meeting, i can do some tests to confirm it.

MarineChap, 2022-05-13:
About the shank number, from the code ([L1288](https://github.com/open-ephys-plugins/neuropixels-pxi/blob/de8abc9b531b1f469ed9fb442439bbd824cc5ee2/Source/UI/NeuropixInterface.cpp#L1288)), it is indeed the second number in CH0="0:0" that gives you the shank. 

```       
        if (probe->type == ProbeType::NP2_4)
               chString += ":" + String(shank);

        channelNode->setAttribute("CH" + String(channel), chString);
        xposNode->setAttribute("CH" + String(channel), String(probe->electrodeMetadata[elec].xpos + 250 * shank));
        yposNode->setAttribute("CH" + String(channel), String(probe->electrodeMetadata[elec].ypos + 250 * shank));
    
```

Be careful there of the PosX and PosY with multiple shanks,  I guess the 250 * shank would be a problem for you no? 

But that also means there is a problem in Open-Ephys side as our multishank imro file does match the setting.xml obtained. 
What do you think ? Does it makes sense ?

alejoe91, 2022-05-13:
Thanks for pointing to the code! I'll implement the same logic into probeinterface making sure the shank info and 250 shift!

@jsiegle is it possible that the shank id is not correctly saved to the settings.xml? There seems to be a mismatch between the imro and the OE settings.

jsiegle, 2022-05-13:
The shank index is written properly with the latest version of the Neuropixels plugin (example file [here](https://github.com/SpikeInterface/probeinterface/files/8690808/settings.txt)). We're currently investigating why the correct shank indices didn't get stored in Marine's case.

alejoe91, 2022-05-16:
And shank ids loaded here: https://github.com/SpikeInterface/probeinterface/pull/107 (this is the data from @jsiegle)

[image]

closed 2022-05-16 by samuelgarcia

MarineChap, 2022-05-16:
Thanks for the fix !