SpikeInterface/probeinterface #110: Limited channel_id - contact_id mapping

JuliaSprenger, 2022-05-27:
Currently probeinterface is expecting a 1-1 match of channel_ids and contact_ids. However in some cases multiple channels were recorded from a single contact (e.g. different frequency ranges). Is there a workaround from the probeinterface side. Do you recommend using a 2nd fake probe to link to the additional channel_ids?

samuelgarcia, 2022-05-30:
If you use probeinterface from spikeinterface, you always have a unique `channel_id` for a given recording which is a unique stream. So in that case the problem do not occurs.

In spikeinterface multi stream format like blackrock must be loaded as separate objects.

probeinterface do not have the concept of `channel_id` we have `contact_id` and for the wiring : `channel_device_index`.
In short when you want to link  a probe and a channel set of a stream this is done with the **indices** and not the **ids** because probeinterface is not aware of the recording.

alejoe91, 2023-08-23:
Seconding @samuelgarcia 

From the probeinterface prospective, you'd need to load two different probes. 

@JuliaSprenger let's move the discussion here: https://github.com/SpikeInterface/ndx-probeinterface/issues/24

closed 2023-08-23 by alejoe91