SpikeInterface/probeinterface #111: to_numpy conflict with spikeglx recordings saving only a subset of channels.

DayanZc, 2022-05-31:
Hi,
I am currently trying to read spikeGLX data with spikeinterface, from a Neuropixels 1.0 probe with a number of saved channels <384. This seems to cause an issue in probeinterface probe.to_numpy as probeinterface.read_spikeglx() will determines the number of contacts by reading the 'len(meta['snsShankMap'])' which is the number of saved channels <384. However, the 'contact_ids' is of size 384 no matter what.

Thus, the Traceback calls : 
     in to_numpy
arr['contact_ids'] = self.contact_ids
ValueError : could not broadcast input array from shape (384,) into shape (200,)

Is changing the metadata file the only way to handle this issue ? Reading data with BinaryExtractor then calling 'set_probe()' raises the same issue.

samuelgarcia, 2022-05-31:
Hi,
Thanks for the report.
In short, you saved 200 channels in the bins but the meta have 384 channels ?
Could you send us the meta file ?
Where is the info of which channel are saved ?

DayanZc, 2022-06-01:
Hello,
Worse than that : the meta file has some entries with 384 channels (the imro table defining the gains, connected bank of each electrode, etc.) whereas the channel map and the shank map only have 200 channels (well 201 actually but it holds true).
The info is stored under the "nSavedChannels" parameter.
Here is the meta file as a .txt for GitHub support
[Day_1_g0_t0.imec0.ap.meta.txt](https://github.com/SpikeInterface/probeinterface/files/8812083/Day_1_g0_t0.imec0.ap.meta.txt)
.

alejoe91, 2023-08-23:
@DayanZc this has been fixed both for SpikeGLX and Open Ephys ;)

- https://github.com/SpikeInterface/probeinterface/pull/180
- https://github.com/SpikeInterface/probeinterface/pull/151

closed 2023-08-23 by alejoe91