SpikeInterface/probeinterface #112: Trouble loading probe file in SI

nasnass, 2022-06-24:
so I have the following probe file (named probe1.prb) for a single electrode recording with one channel. 
total_nb_channels = 15000
radius            = 250
channel_groups    = {
	1 : {
		'channels': list(range(1)),
		'graph' : [],
		'geometry' :{
			0 :[ 0,0],
		}
	}
}

I've tried running: 
probe = pi.read_prb('probe1.prb') to load the file 

I then try to save it so that I can pass it to the sorter (in this case herding spikes) :
recording_prb = data.Probe(probe) and get the following error:

AttributeError                            Traceback (most recent call last)
Input In [44], in <cell line: 1>()
----> 1 recording_prb = data.Probe(probe)

AttributeError: 'File' object has no attribute 'Probe'

I have tried the following as well:

recording_prb = data.pi.read_prb("probe1.prb") and get the same error message. Any help would be appreciated!

alejoe91, 2022-06-24:
Hi @nasnass 

How are you saving the probe? What is `data`? Can you paste your script?

alejoe91, 2022-06-24:
Also note that herding spikes is not designed for single channels!

nasnass, 2022-06-24:
Thank you for the hint about herding spikes! I am also trying to sort the data using spyking circus. 

Data is the data that is being used -hdf file which is read in using : data = h5py.File(filename, 'r')

I think saving the probe is exactly my problem, there was another issue open on this matter, however I tried the steps there and it didn't resolve my problem. 

probe = pi.read_prb('probe1.prb') ##reading probe
data_prb = data.Probe(probe)  ##setting the probe to the data file

alejoe91, 2022-06-24:
Ok, so the issue is that `data` is an `h5py.File` object and it has no `Probe` field. What you get from reading with probeinterface in addition is a `probeinterface.Probe` object, which cannot be added to an h5 file. I'm not sure exactly what you're trying to do. Are you using spikeinterface to run spike sorting? 

In that case, I would suggest you use either WaveClus or combinato as sorters, as they where designed for single-channel recordings. Also, for these kind of recordings you don't really need a probe file, because it doesn't add any information (geometry is needed when the sorters use it as an additional information). WaveClus and combinato don't need it.

closed 2022-06-29 by nasnass