SpikeInterface/probeinterface #113: Load Open-ephys Neuropixels PXI recording with sync channel appended

LiesDec, 2022-06-29:
The Neuropixels PXI plugin has the option to append the **sync line **  to the data stream as a continuous channel (see [documentation](https://open-ephys.github.io/gui-docs/User-Manual/Plugins/Neuropixels-PXI.html#synchronization)). That means that the `continuous.dat` file now has one channel more than is specified in the probe settings. This causes an error when loading the files as there is a channel count  check done before loading ([#L415](https://github.com/SpikeInterface/spikeinterface/blob/master/spikeinterface/core/baserecording.py#L415)) because if the probe is already defined `set_probe(inplace=True)`, (which is the case when using the Neuropixels PXI recordings) it assumes the the number of recorded channels must be equal to the number of electrodes on the probe.

>         if in_place:
>             if not np.array_equal(new_channel_ids, self.get_channel_ids()):
>                 raise Exception('set_proce(inplace=True) must have all channel indices')
>             sub_recording = self
>         else:
>             if np.array_equal(new_channel_ids, self.get_channel_ids()):
>                 sub_recording = self.clone()
>             else:
>                 sub_recording = ChannelSliceRecording(self, new_channel_ids)
>       
>         sub_recording.set_property('contact_vector', arr, ids=None)

This is a tricky issue (I think) as it comes down to having to change the way the recording is read using the open-ephys extractors. We should do something similar as is done with the spikeglx data, that is checking for a sync channel and separating it off early on. (I think it is done here [#L120](https://github.com/NeuralEnsemble/python-neo/blob/master/neo/rawio/spikeglxrawio.py#L120)

alejoe91, 2022-06-29:
Thank you for the issue! Can you share a small test file with the `sync` channel enabled? Ideally, something super short would be great (like 1s) and it doesn't need to have real data. We use these kind of data in [this repo](https://gin.g-node.org/NeuralEnsemble/ephy_testing_data) to test multiple cases for different acquisition devices :)

alejoe91, 2022-06-29:
@LiesDec you can send it to us via email and we'll take care of uploading it :) My email in the [docs](https://spikeinterface.readthedocs.io/en/latest/authors.html). Also send us the details of who made the recording and in which lab so we'll add those info as "acknowledgments"

samuelgarcia, 2022-06-29:
No the blacrock sortig is not ported. But if you need it this could be done with a wrapper.

LiesDec, 2022-07-01:
@samuelgarcia Great that you're already on this issue. Do you still need a dataset then? :)

Also, I noticed that I only have the loading error when I have only one neuropixel 2.0 attached to the headstone (one data stream). When I have two and the sync channel appended as continuous channel I do not have the reading error (despite if for sure being registered).  Maybe this is related to this other issue I came across? https://github.com/SpikeInterface/probeinterface/issues/114

For reference, these are the two `settings.xml` files for a recording with one and two probes attached but both with the sync channel recorded a continuous channel

- 1 probe, Sync channel appended and error loading
[settings_1p.txt](https://github.com/SpikeInterface/probeinterface/files/9028675/settings_1p.txt)

- 2 probes, Sync channel appended and no error loading
[settings_2p.txt](https://github.com/SpikeInterface/probeinterface/files/9028677/settings_2p.txt)

Aryo-Zare, 2022-07-05:
Is the dataset needed and already uploaded ? 
If not, I activated sync channel in all of my recordings. Hence I may provide my recordings.

samuelgarcia, 2022-07-05:
We already have the dataset with sync I think

alejoe91, 2023-08-23:
This is fixed on the SpikeInterface side.

To summarize, the probe will always load the 384 channels (or less is a subset is used) and it will be attached to a `recording` only if `load_sync_channel` is False. This is to prevent a wrong usage of the probes when the sync channel is loaded. See https://github.com/SpikeInterface/spikeinterface/pull/1830

closed 2023-08-23 by alejoe91