SpikeInterface/probeinterface #114: Neuropixels PXI probe info loading two probes attached

LiesDec, 2022-07-01:
There is an error in loading the channel information when having two probes attached to one headstage with the neuropixels 2.0.
In loading the recording you immediately specify which data stream you want to read (0, or 1) which corresponds to the two different probes attached to the same headstage

```
 import spikeinterface.extractors as se  
 recording_0 = se.read_openephys(Path/to/data/folder, stream_id='0')
```

When reading the channel info for setting the probe, the data stream is not taken into account (specified in the settings.xml file with 'dock'). Hence the channelmap of dock 0 is applied to both data streams (but these are two different probes and thus have different channel-maps). See [#L882 to #L890 ](https://github.com/SpikeInterface/probeinterface/blob/main/probeinterface/io.py#L882-L890)

 ```
for child in npix:
        if child.tag == "EDITOR":
            for child2 in child:
                if child2.tag == "NP_PROBE":
                    probe_name = child2.attrib["probe_name"]
                    probe_serial_number = child2.attrib["probe_serial_number"]
                    probe_part_number = child2.attrib["probe_part_number"]
                    np_probe = child2
                    break
```

If I change the code to this, both separate channel-maps are correctly loaded

```
 for child in npix:
        if child.tag == "EDITOR":
            for child2 in child:
                if child2.tag == "NP_PROBE":
                    if child2.attrib["dock"]==str(int(stream_id)+1):
                        probe_name = child2.attrib["probe_name"]
                        probe_serial_number = child2.attrib["probe_serial_number"]
                        probe_part_number = child2.attrib["probe_part_number"]
                        np_probe = child2
                        break
```

No sure why this is named 'dock' in the `settings.xml` file and I'm also unsure if this code change would impact reading other data recorded with the Neuropixel PXI extractor. Hence an issue rather than I pull request :)

LiesDec, 2022-07-04:
@alejoe91 Shall I try to create a pull request for this?

alejoe91, 2022-07-04:
Hi @LiesDec 

Actually yes, I'm a bitt busy these days! Thank you!

MarineChap, 2022-07-26:
@alejoe91 Hello, I am doing the follow-up for Lies on this issue. I can do the PR but I am wondering how you think we should proceed to select which channelmap we want.  If I return a probegroup containing both probes, open-ephys will be able to handle it correctly and select the right one based on the stream_id ? That's the way to go  or you would prefer that we add a stream_id parameters to select the probe we want to return ?

LiesDec, 2022-08-04:
@alejoe91 
I found another issue regarding the loading the probe data of neuropixel data with open-ephys. 
The electrode `positions` and  `channel_names` are loaded in the same order as they are written in the `settings.xml` file.
see [#L896-L915](https://github.com/SpikeInterface/probeinterface/blob/main/probeinterface/io.py#L896-L915)
```  
# read channels
    for child in np_probe:
        if child.tag == "CHANNELS":
            channel_names = np.array(list(child.attrib.keys()))
            channel_values = np.array(list(child.attrib.values()))
            # check if shank ids is present
            if all(":" in val for val in channel_values):
                shank_ids = [int(val[val.find(":") + 1:]) for val in channel_values]
            else:
                shank_ids = None
# read positions
    for child in np_probe:
        if child.tag == "ELECTRODE_XPOS":
            xpos = [float(child.attrib[ch]) for ch in channel_names]
        if child.tag == "ELECTRODE_YPOS":
            ypos = [float(child.attrib[ch]) for ch in channel_names]
    positions = np.array([xpos, ypos]).T
```

In later code,  it is assumed that the electrode positions are sorted, that is electrode position 1 corresponds to 'CH1' and electrode position 2 corresponds to 'CH2'. 
for example the plotting of the channel indices on the probe (see [#L148-166 ](https://github.com/SpikeInterface/probeinterface/blob/main/probeinterface/plotting.py#L148-L166)),
```
if with_channel_index or with_contact_id or with_device_index or text_on_contact is not None:
        if probe.ndim == 3:
            raise NotImplementedError('Channel index is 2d only')
        for i in range(n):
            txt = []
            if with_channel_index:
                txt.append(f'{i + first_index}')
            if with_contact_id and probe.contact_ids is not None:
                contact_id = probe.contact_ids[i]
                txt.append(f'id{contact_id}')
            if with_device_index and probe.device_channel_indices is not None:
                chan_ind = probe.device_channel_indices[i]
                txt.append(f'dev{chan_ind}')
            if text_on_contact is not None:
                txt.append(f'{text_on_contact[i]}')
            
            txt = '\n'.join(txt)
            x, y = probe.contact_positions[i]
            ax.text(x, y, txt, ha='center', va='center', clip_on=True)
```
However, this assumed sorting is not always the case (for example not in my case). 

It would be good to first sort the channel_names before loading the electrode positions. 
This is how I did it and it worked, but feel free to find a different/better way :)

```
    # read channels
    for child in np_probe:
        if child.tag == "CHANNELS":
            channel_names = np.array(list(child.attrib.keys()))
            channel_values = np.array(list(child.attrib.values()))
            # check if shank ids is present
            if all(":" in val for val in channel_values):
                shank_ids = [int(val[val.find(":") + 1:]) for val in channel_values]
            else:
                shank_ids = None

    channel_names_number = [int(cn[2:]) for cn in channel_names]
    channel_names = np.array(['CH'+str(scn) for scn in np.sort(channel_names_number)])
    channel_values = np.array([x for _,x in sorted(zip(channel_names_number,channel_values))] )
    shank_ids = [x for _,x in sorted(zip(channel_names_number,shank_ids))]  
```
p.s @MarineChap this solves our channel map problem I think :). However, It would be good that I can check this with a `settings.xml` file form a recording with the new versions of the Neuropixels_PXI plugin. Do you still have that recording/`settings.xml` file?

LiesDec, 2022-08-04:
Hi Samuel,

I'm on a little holiday over the weekend from Friday at noon. But we could schedule something next week on Monday, 2.30pm CEST?

alejoe91, 2022-08-05:
> I'm on a little holiday over the weekend from Friday at noon. But we could schedule something next week on Monday, 2.30pm CEST?

Sounds good! :) I'll send you an email!

alejoe91, 2023-04-03:
This is implemented

closed 2023-04-03 by alejoe91