SpikeInterface/probeinterface #115: Channel Editor?

AntonioST, 2022-07-15:
Do you have any plan for providing the channel editor, such as the channel editor in Open Ephys? For now, I am using the 4-shank Neuropixels, and I found the channel editor in Open Ephys is so simple that it is hard to understand the rule behind the Neuropixels.
For example, only one electrode can be selected as readout channel from all electrodes shared with same channel ID. I cannot easy finding out which channels will be disabled when I enable certain channels. For an another example, according to my experiment, I want to recording multiple brain regions along the probe. However, I cannot assign full density channels around those regions, so I used the half density arrangement by skip some rows. By the channel editor in Open Ephys, I need to click the channels one by one. Although I can write some Python script to do that for me, It is still not convenient. It would be better have a function to automatically pick channels under certain density rule. Especially I also want to plot the probe and channels on the brain slice, either histology image or Atlas mouse reference brain. It would be great that I have a brain reference to know where the channels are when I am picking the channels.

samuelgarcia, 2022-07-18:
Hi Antonio,
I am not sure to follow you.
The purpose of probeinterface is to read several probe description format and unify then with a simple object model.

I never used the OpenEphys channel editor. So I can't tell anything.
What I guess is that it is a tool to select 384 channel among 1024.
The NP probe have many constrain on how channel can be selected because of adc groups.
Such an editor/selector is really related to the NP probe (so spikeglx or openephys)

Also note that probeinterface do not have ability to write to any format except the internal one (json based).

I would be happy to have more discussion on this if you want.

alejoe91, 2023-08-23:
@AntonioST `probeinterface` can only load files and it cannot be used to select NP channels. Closing

closed 2023-08-23 by alejoe91