SpikeInterface/probeinterface #133: All top-level directories installed to site-packages

musicinmybrain, 2022-10-10:
Since the package was converted to `pyproject.toml` for 0.2.12, there are things in the `site-packages` directory that should not be there:

```
$ python3 -m venv _e
$ . _e/bin/activate
(_e) $ pip install probeinterface
[…]
Successfully installed numpy-1.23.3 probeinterface-0.2.12
[…]
(_e) $ ls -1 _e/lib/python3.10/site-packages/
_distutils_hack
distutils-precedence.pth
doc
examples
numpy
numpy-1.23.3.dist-info
numpy.libs
pip
pip-21.3.1.dist-info
pkg_resources
probeinterface
probeinterface-0.2.12.dist-info
resources
setuptools
setuptools-59.6.0.dist-info
tests
```

Note that the top-level `doc`, `examples`, `resources`, and `tests` directories from `probeinterface` have been installed into the virtualenv-wide (or system-wide, if installed that way) `site-packages`.

musicinmybrain, 2022-10-10:
I’ve offered one candidate fix in https://github.com/SpikeInterface/probeinterface/pull/134.

h-mayorquin, 2023-06-23:
Was this solved for you after https://github.com/SpikeInterface/probeinterface/pull/136?

Maybe we should close

musicinmybrain, 2023-06-27:
> Was this solved for you after #136?
> 
> Maybe we should close

I agree, this is fixed. Thanks.