SpikeInterface/probeinterface #141: incorrect wiring with io.read_spikeglx ?

TomBugnon, 2022-10-21:
Hi, 

Sorry if I'm getting confused again with probe wiring, but `probeinterface.io.read_spikeglx(metapath)` doesn't seem to reflect user channel order properly.

Here is what I get with 
```
rec_prb = rec.set_probe(read_spikeglx(metapath))

sw.plot_timeseries(
    rec_prb,
    segment_index=0,
    order_channel_by_depth=True,
    time_range=(0, 0.05)
)
```

[image]

It is the same plot with `order_channel_by_depth=False`

and here is `prb.to_dataframe()`
[image]

is [this](https://github.com/SpikeInterface/probeinterface/blob/6ef3a121fecce11473fe3d93af1e5af3a8d9ba59/probeinterface/io.py#L897) line correct?

Thanks

samuelgarcia, 2022-10-21:
Hi.
For spikeglx, channel_ids and contact_ids are in the same order because organise in the same the same.
There is no complex "wiring" there contrary to neuronexus+intan+openephys for instance.

I guess that in your config channel are already order by depth naturaly. So odering for display do not play a role.

TomBugnon, 2022-10-25:
Hi,
Thanks for your response.
Sorry, I don't understand. Initializing the probe with this method gives different results than setting directly x and y coordinates for each channel.

Here's what the probe, contact locations recording look like with `read_spikeglx(metapath)`. It says channels span the first 3000um although this probe was configured such that channels actually span 7600um:
[image]
[image]
[image]

So far, the way we have been obtaining x and y coordinates for each channel was with [this](https://github.com/CSC-UW/ecephys/blob/develop/ecephys/sglx/external/SGLXMetaToCoords.py) "SGLXMetatoCoords" module (slightly modified [from Jennifer Colonel's](https://github.com/jenniferColonell/ecephys_spike_sorting/blob/master/ecephys_spike_sorting/common/SGLXMetaToCoords.py) to be able to return x and y coords.)
Now if we set the contact positions of the probe directly from the channel xy coordinates:
[image]
Now the probe looks correct (checkerBox pattern):
[image]
And the recording orders by depth properly:
[image]

samuelgarcia, 2022-10-25:
Hi Tom.
This is really really annoying!!
This would means that we have a strong bug in the way we are reading the channel location from the meta file.

I have to admit that I do not have too much knowledge about that

@MarineChap @jenniferColonell: 
could you coordinate to compare how we are reading the meta file between the 2 repos:
https://github.com/jenniferColonell/ecephys_spike_sorting/blob/master/ecephys_spike_sorting/common/SGLXMetaToCoords.py
https://github.com/SpikeInterface/probeinterface/blob/main/probeinterface/io.py#L939

My intuition is that there is a mismatch between the channel_id and the contact_id for the formula of the x/y position.

samuelgarcia, 2022-10-25:
@TomBugnon : could you send us your meta file so we could include it in our tests suite ?

TomBugnon, 2022-10-25:
Sorry if I'm a bearer of bad news here.

Also, now I am noticing: in `plot_timeseries()` the y axis seems to be inverted relative to how `phy` shows depths (in phy: 0 = tip of probe. Here it looks like 0=furthest from tip). In this recording most spikes are in the thalamus, towards the tip of the electrode.  This is what the recording looks like in Phy (with channel_map obtained as I said previously). 
[image]

[image]

TomBugnon, 2022-10-25:
Yep looks correct with 0.96.0, thanks
[image]

@samuelgarcia  
https://uwmadison.box.com/s/afo1h104392dnd2md561e3bxsq622p6t

jenniferColonell, 2022-10-25:
Hi guys,

So, is the incorrect geometry problem solved with the newer version of SpikeInterface? Or are there still some differences in how Spikeinterface is interpreting metadata that need to be resolved?

thanks!

jennifer

External Email: Use Caution

[image]<https://urldefense.com/v3/__https://user-images.githubusercontent.com/14128078/197821509-97d94038-9226-4720-ba2b-202b326dd21b.png__;!!Eh6p8Q!HdVp_CxiHvT_hmCHyhPnhTgvVQtMsqk1Zka2zTdfz3Fo5Nh4d4UWIZjlBoYp8YDej3NzpI9QWhP0h5-YZSPONWL9UJv_dQ$>

https://uwmadison.box.com/s/afo1h104392dnd2md561e3bxsq622p6t<https://urldefense.com/v3/__https://uwmadison.box.com/s/afo1h104392dnd2md561e3bxsq622p6t__;!!Eh6p8Q!HdVp_CxiHvT_hmCHyhPnhTgvVQtMsqk1Zka2zTdfz3Fo5Nh4d4UWIZjlBoYp8YDej3NzpI9QWhP0h5-YZSPONWL7j-fpIQ$>

External Email: Use Caution

[image]<https://urldefense.com/v3/__https://user-images.githubusercontent.com/14128078/197821509-97d94038-9226-4720-ba2b-202b326dd21b.png__;!!Eh6p8Q!HdVp_CxiHvT_hmCHyhPnhTgvVQtMsqk1Zka2zTdfz3Fo5Nh4d4UWIZjlBoYp8YDej3NzpI9QWhP0h5-YZSPONWL9UJv_dQ$>

https://uwmadison.box.com/s/afo1h104392dnd2md561e3bxsq622p6t<https://urldefense.com/v3/__https://uwmadison.box.com/s/afo1h104392dnd2md561e3bxsq622p6t__;!!Eh6p8Q!HdVp_CxiHvT_hmCHyhPnhTgvVQtMsqk1Zka2zTdfz3Fo5Nh4d4UWIZjlBoYp8YDej3NzpI9QWhP0h5-YZSPONWL7j-fpIQ$>

samuelgarcia, 2022-10-25:
@jenniferColonell : no the new version only solve the flip top/down of the image of time series.

The channel map diffrerences are unsolved.

samuelgarcia, 2022-10-25:
@MarineChap: could b ethat this line should use `elec_id` instead of  `channel_id`
https://github.com/SpikeInterface/probeinterface/blob/main/probeinterface/io.py#L839

Or maybe this the very recent patch by @rtraghavan:
https://github.com/SpikeInterface/probeinterface/pull/126
https://github.com/SpikeInterface/probeinterface/blob/main/probeinterface/io.py#L806

samuelgarcia, 2022-10-25:
@TomBugnon : could your try to get the actual master and then to comment line from 806 to 829 in io.py.
And tell us if it fix the problem ?

alejoe91, 2022-10-25:
thank you @TomBugnon 

This is really helpful, we'll do more tests on our side and hopefully fix this by the end of the week!

alejoe91, 2022-10-25:
@TomBugnon can you test this? https://github.com/SpikeInterface/probeinterface/pull/142

TomBugnon, 2022-10-25:
@alejoe91  Looks good with #142  !

The probe map is correct, the recording shows properly.
Channel X and Y coordinates are identical except for a translation along X-axis (MetatoCoord takes probe edge as origin, read_sglx takes lower-left channel I believe)
[image]

Thank you!

grahamfindlay, 2022-10-25:
Hi all. In case it is helpful, I have an ImecMap class that uses Jennifer's code to parse the Neuropixel 1.0 Imec Readout Table (imro) and Imec Channel Map (cmp) directly from a .bin file's metadata (or other sources) and, and puts together a handy table that keeps straight the user order, channel order, electrode site, channel id, x and y coordinates, etc. We use funky probe layouts all the time, and so far haven't encountered any issues. [Link](https://github.com/CSC-UW/ecephys/tree/develop/ecephys/sglxr). 

[image]

alejoe91, 2022-10-25:
Awesome! I guess we only have test data with standard configs so we never spotted this. I'll double check that everything is the same also for NP2 and include your files for testing, then we'll merge and make a new release. Sorry about the hassle!

jenniferColonell, 2022-10-25:
Site to channel mapping for NP2 is significantly different than NP1.0 – which I assume you know, but just in case!
My code is correct as far as I know, and the IBL code using the shank map also works.

alejoe91, 2022-10-25:
Thanks @jenniferColonell 

Do you happen to have some test meta files for now probes, also with non-standard configurations?

jenniferColonell, 2022-10-25:
Hi Alessio,

I’ll generate a few for you.

jennifer

jenniferColonell, 2022-10-25:
OK, here’s a little variety of 2.0 metadata. They are probably wackier patterns than most experiments would use – but that’s the point.

I have to say: although my code translates these using the imro table, its actually more efficient to use the snsShankMap. My code suffers from my history of first making a routine to create the imro tables (and I should fix that). While imro table entries are different for different probe types, snsShankMap entries are the same, (shank:column:row:enabled).

At some point I’ll switch my code over, but there’s only 24 hours in a day!

alejoe91, 2022-10-25:
Thanks @jenniferColonell 

I don't see the attachments though. Could you send them over on slack?

jenniferColonell, 2022-10-25:
Sure, I’ll send you a dm.

jenniferColonell, 2022-10-25:
Should have specified: I sent the attachments via the Neuropixels slack.

samuelgarcia, 2022-10-26:
THanks all for the help and feeback.
Alessio find the bug.
@jenniferColonell : do you allow us to push publically (in the repo) some of your files for running CI tests ?

jenniferColonell, 2022-10-26:
No problem about pushing the code, with attribution, but which code is it?

alejoe91, 2022-10-27:
Fix released in [0.2.14](https://pypi.org/project/probeinterface/0.2.14/) and release notes here: https://probeinterface.readthedocs.io/en/main/releases/0.2.14.html

Note that the bug was introduced on Sept. 1st 2022, so not too long ago.

Thank you all for reporting it, providing test data, and support!

closed 2022-10-27 by alejoe91