SpikeInterface/probeinterface #145: pypi archive doesn't package tests

TheChymera, 2022-11-03:
I generally prefer to reference pypi archives for packaging under gentoo, mainly because some developers have the habit of retroactively “fixing” releases on github, which leads to checksum mismatch and packages needing to be re-tested. While this hasn't happened with probeinterfaces, it would be cool if we could use pypi.

Sadly the archive does not include the test suite:

```console
chymera@decohost /tmp/probeinterf $ wget https://files.pythonhosted.org/packages/50/6c/03265ae36d98238e3211be3410c5f817a0e721f159d0e355fcadaf27f558/probeinterface-0.2.14.tar.gz
--2022-11-03 15:48:08--  https://files.pythonhosted.org/packages/50/6c/03265ae36d98238e3211be3410c5f817a0e721f159d0e355fcadaf27f558/probeinterface-0.2.14.tar.gz
Resolving files.pythonhosted.org... 2a04:4e42:600::319, 2a04:4e42::319, 2a04:4e42:200::319, ...
Connecting to files.pythonhosted.org|2a04:4e42:600::319|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 30767 (30K) [application/x-tar]
Saving to: ‘probeinterface-0.2.14.tar.gz’

probeinterface-0.2.14.tar.gz       100%[================================================================>]  30.05K  --.-KB/s    in 0.02s

2022-11-03 15:48:08 (1.21 MB/s) - ‘probeinterface-0.2.14.tar.gz’ saved [30767/30767]

chymera@decohost /tmp/probeinterf $ tar xvf probeinterface-0.2.14.tar.gz
probeinterface-0.2.14/
probeinterface-0.2.14/LICENSE
probeinterface-0.2.14/PKG-INFO
probeinterface-0.2.14/README.md
probeinterface-0.2.14/probeinterface/
probeinterface-0.2.14/probeinterface/__init__.py
probeinterface-0.2.14/probeinterface/generator.py
probeinterface-0.2.14/probeinterface/io.py
probeinterface-0.2.14/probeinterface/library.py
probeinterface-0.2.14/probeinterface/plotting.py
probeinterface-0.2.14/probeinterface/probe.py
probeinterface-0.2.14/probeinterface/probegroup.py
probeinterface-0.2.14/probeinterface/shank.py
probeinterface-0.2.14/probeinterface/utils.py
probeinterface-0.2.14/probeinterface/wiring.py
probeinterface-0.2.14/probeinterface.egg-info/
probeinterface-0.2.14/probeinterface.egg-info/PKG-INFO
probeinterface-0.2.14/probeinterface.egg-info/SOURCES.txt
probeinterface-0.2.14/probeinterface.egg-info/dependency_links.txt
probeinterface-0.2.14/probeinterface.egg-info/requires.txt
probeinterface-0.2.14/probeinterface.egg-info/top_level.txt
probeinterface-0.2.14/pyproject.toml
probeinterface-0.2.14/setup.cfg
probeinterface-0.2.14/setup.py
chymera@decohost /tmp/probeinterf $ mv probeinterface-0.2.14 pypi
chymera@decohost /tmp/probeinterf $ rm probeinterface-0.2.14.tar.gz
chymera@decohost /tmp/probeinterf $ wget https://github.com/SpikeInterface/probeinterface/archive/refs/tags/0.2.14.tar.gz
--2022-11-03 15:48:52--  https://github.com/SpikeInterface/probeinterface/archive/refs/tags/0.2.14.tar.gz
Resolving github.com... 140.82.112.4
Connecting to github.com|140.82.112.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/SpikeInterface/probeinterface/tar.gz/refs/tags/0.2.14 [following]
--2022-11-03 15:48:52--  https://codeload.github.com/SpikeInterface/probeinterface/tar.gz/refs/tags/0.2.14
Resolving codeload.github.com... 140.82.112.9
Connecting to codeload.github.com|140.82.112.9|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/x-gzip]
Saving to: ‘0.2.14.tar.gz’

0.2.14.tar.gz                          [ <=>                                                             ] 205.30K  --.-KB/s    in 0.1s

2022-11-03 15:48:53 (1.58 MB/s) - ‘0.2.14.tar.gz’ saved [210232]

chymera@decohost /tmp/probeinterf $ tar xvf 0.2.14.tar.gz
probeinterface-0.2.14/
probeinterface-0.2.14/.github/
probeinterface-0.2.14/.github/workflows/
probeinterface-0.2.14/.github/workflows/publish-to-pypi.yml
probeinterface-0.2.14/.github/workflows/python-package.yml
probeinterface-0.2.14/.gitignore
probeinterface-0.2.14/.readthedocs.yaml
probeinterface-0.2.14/LICENSE
probeinterface-0.2.14/README.md
probeinterface-0.2.14/doc/
probeinterface-0.2.14/doc/Makefile
probeinterface-0.2.14/doc/api.rst
probeinterface-0.2.14/doc/conf.py
probeinterface-0.2.14/doc/format_spec.rst
probeinterface-0.2.14/doc/generate_format_example.py
probeinterface-0.2.14/doc/img/
probeinterface-0.2.14/doc/img/probe_format_example.png
probeinterface-0.2.14/doc/img/probeinterface_naming.png
probeinterface-0.2.14/doc/index.rst
probeinterface-0.2.14/doc/library.rst
probeinterface-0.2.14/doc/make.bat
probeinterface-0.2.14/doc/overview.rst
probeinterface-0.2.14/doc/probe_format_example.json
probeinterface-0.2.14/doc/release_notes.rst
probeinterface-0.2.14/doc/releases/
probeinterface-0.2.14/doc/releases/0.1.0.rst
probeinterface-0.2.14/doc/releases/0.2.0.rst
probeinterface-0.2.14/doc/releases/0.2.1.rst
probeinterface-0.2.14/doc/releases/0.2.10.rst
probeinterface-0.2.14/doc/releases/0.2.11.rst
probeinterface-0.2.14/doc/releases/0.2.12.rst
probeinterface-0.2.14/doc/releases/0.2.13.rst
probeinterface-0.2.14/doc/releases/0.2.14.rst
probeinterface-0.2.14/doc/releases/0.2.2.rst
probeinterface-0.2.14/doc/releases/0.2.3.rst
probeinterface-0.2.14/doc/releases/0.2.4.rst
probeinterface-0.2.14/doc/releases/0.2.5.rst
probeinterface-0.2.14/doc/releases/0.2.6.rst
probeinterface-0.2.14/doc/releases/0.2.7.rst
probeinterface-0.2.14/doc/releases/0.2.8.rst
probeinterface-0.2.14/doc/releases/0.2.9.rst
probeinterface-0.2.14/examples/
probeinterface-0.2.14/examples/README.rst
probeinterface-0.2.14/examples/ex_01_generate_probe_from_sratch.py
probeinterface-0.2.14/examples/ex_02_probe_2d_probe_3d.py
probeinterface-0.2.14/examples/ex_03_generate_probe_group.py
probeinterface-0.2.14/examples/ex_04_multi_shank_probe.py
probeinterface-0.2.14/examples/ex_05_device_channel_indices.py
probeinterface-0.2.14/examples/ex_06_import_export_to_file.py
probeinterface-0.2.14/examples/ex_07_probe_generator.py
probeinterface-0.2.14/examples/ex_08_more_plotting_examples.py
probeinterface-0.2.14/examples/ex_09_more_complicated_probe.py
probeinterface-0.2.14/examples/ex_10_get_probe_from_library.py
probeinterface-0.2.14/examples/ex_11_automatic_wiring.py
probeinterface-0.2.14/examples/ex_12_plot_values.py
probeinterface-0.2.14/probeinterface/
probeinterface-0.2.14/probeinterface/__init__.py
probeinterface-0.2.14/probeinterface/generator.py
probeinterface-0.2.14/probeinterface/io.py
probeinterface-0.2.14/probeinterface/library.py
probeinterface-0.2.14/probeinterface/plotting.py
probeinterface-0.2.14/probeinterface/probe.py
probeinterface-0.2.14/probeinterface/probegroup.py
probeinterface-0.2.14/probeinterface/shank.py
probeinterface-0.2.14/probeinterface/utils.py
probeinterface-0.2.14/probeinterface/wiring.py
probeinterface-0.2.14/pyproject.toml
probeinterface-0.2.14/resources/
probeinterface-0.2.14/resources/generate_cambridgeneurotech_libray.py
probeinterface-0.2.14/setup.py
probeinterface-0.2.14/tests/
probeinterface-0.2.14/tests/NP24_g0_t0.imec0.ap.meta
probeinterface-0.2.14/tests/NeuropixelPhase3A_file_g0_t0.imec.ap.meta
probeinterface-0.2.14/tests/Noise_g0_t0.imec0.ap.meta
probeinterface-0.2.14/tests/OE_Neuropix-PXI-multi-probe/
probeinterface-0.2.14/tests/OE_Neuropix-PXI-multi-probe/settings.xml
probeinterface-0.2.14/tests/OE_Neuropix-PXI/
probeinterface-0.2.14/tests/OE_Neuropix-PXI/settings.xml
probeinterface-0.2.14/tests/TEST_20210920_0_g0_t0.imec0.ap.meta
probeinterface-0.2.14/tests/allan-longcol_g0_t0.imec0.ap.meta
probeinterface-0.2.14/tests/doppio-checkerboard_t0.imec0.ap.meta
probeinterface-0.2.14/tests/dummy.prb
probeinterface-0.2.14/tests/p2_g0_t0.imec0.ap.meta
probeinterface-0.2.14/tests/test_generator.py
probeinterface-0.2.14/tests/test_io.py
probeinterface-0.2.14/tests/test_library.py
probeinterface-0.2.14/tests/test_multi_shank.imro
probeinterface-0.2.14/tests/test_plotting.py
probeinterface-0.2.14/tests/test_probe.py
probeinterface-0.2.14/tests/test_probegroup.py
probeinterface-0.2.14/tests/test_wiring.py
chymera@decohost /tmp/probeinterf $ mv probeinterface-0.2.14/ gh
chymera@decohost /tmp/probeinterf $ pushd gh/
/tmp/probeinterf/gh /tmp/probeinterf
chymera@decohost /tmp/probeinterf/gh $ find -name "*test*"
./tests
./tests/test_probe.py
./tests/test_io.py
./tests/test_plotting.py
./tests/test_generator.py
./tests/test_multi_shank.imro
./tests/test_probegroup.py
./tests/test_library.py
./tests/test_wiring.py
chymera@decohost /tmp/probeinterf/gh $ popd
/tmp/probeinterf
chymera@decohost /tmp/probeinterf $ pushd pypi/
/tmp/probeinterf/pypi /tmp/probeinterf
chymera@decohost /tmp/probeinterf/pypi $ find -name "*test*"
chymera@decohost /tmp/probeinterf/pypi $
```

would you be interested in including these files in the pypi archive?

samuelgarcia, 2022-11-09:
Hi Christian.
Contrary to other package (spikeinterface) we choose to put the "tests" folder outside the code folder because it contains some small testing files (txt, meta, txt, prb). This would make the `pip install probeinterface` a bit more heavy.

If you need this folder for gentoo packaging we can had it in the tar.gz.
I think it would be cool to not have it in the wheel. No ?
Maybe there is a way to include the tests folde rin the tar.gz on pypi but not wheel. Any idea how ?

TheChymera, 2022-11-09:
@samuelgarcia yes, that would be great! we do not use the wheel, so `.tar.gz` is the only place where it would actually make a difference :)

musicinmybrain, 2022-12-21:
[Fedora Linux `python-probeinterface` package](https://src.fedoraproject.org/rpms/python-probeinterface) maintainer here. Most of the Python packages I maintain have tests, docs, or something else missing from the PyPI sdists, which drives me to package from GitHub source archives instead. I don’t mind that I need to do so for this package too. That said, I too would favor including the tests in the sdist but not in the wheels.

alejoe91, 2023-08-23:
Tests are part of the `tar.gz`, closing!

closed 2023-08-23 by alejoe91