SpikeInterface/probeinterface #152: Cannot read OpenEphys XML file using read_openephys

himahuja, 2023-01-05:
I am trying to manually load the settings.xml from one of our Neuropixel recordings and I get this error. I am also attaching the XML [file for the same](https://gist.githubusercontent.com/himahuja/c11934f2bce5aefd6d9c2acf6658d920/raw/3426bfb8f828f14e2a487e0e5549442bb86d995b/settings.xml). It appears to be an XML/ Openephys issue. 

`pi.read_openephys('ephys_2022_12_15/ForepawStimDay1/Record Node 102/settings.xml')`
Output:

```
> ---------------------------------------------------------------------------
> InvalidVersion                            Traceback (most recent call last)
> ~\AppData\Local\Temp\ipykernel_4544\2788258613.py in <module>
> ----> 1 pi.read_openephys('ephys_2022_12_15/ForepawStimDay1/Record Node 102/settings.xml')
> 
> ~\anaconda3\envs\nwb7\lib\site-packages\probeinterface\io.py in read_openephys(settings_file, stream_name, probe_name, serial_number, fix_x_position_for_oe_5, raise_error)
>     991 
>     992     info_chain = root.find("INFO")
> --> 993     oe_version = parse(str(info_chain.find("VERSION")))
>     994     signal_chain = root.find("SIGNALCHAIN")
>     995     neuropix_pxi = None
> 
> ~\anaconda3\envs\nwb7\lib\site-packages\packaging\version.py in parse(version)
>      50     :raises InvalidVersion: When the version string is not a valid version.
>      51     """
> ---> 52     return Version(version)
>      53 
>      54 
> 
> ~\anaconda3\envs\nwb7\lib\site-packages\packaging\version.py in __init__(self, version)
>     195         match = self._regex.search(version)
>     196         if not match:
> --> 197             raise InvalidVersion(f"Invalid version: '{version}'")
>     198 
>     199         # Store the parsed out pieces of the version
> 
> InvalidVersion: Invalid version: '<Element 'VERSION' at 0x000002520C86F598>'
```

alejoe91, 2023-01-05:
This is fixed in the latest release. Can you upgrade the package? `pip install -U probeinterface`

himahuja, 2023-01-06:
This worked for me on Python 3.8, probeinterface==0.2.15. Thank you very much!

closed 2023-01-06 by himahuja