SpikeInterface/probeinterface #155: New Neuropixel 1.0 NHP probe geometries not handled properly

JonathanAMichaels, 2023-02-24:
Hello,

It looks like probeinterface hasn't been updated for the newest Neuropixel geometries. There are new NHP 1.0 probes (described here https://www.neuropixels.org/_files/ugd/328966_97dfdcf9421a41ae9c57c9047928834c.pdf)

They are NP 1.0, so they are imDatPrb_type = 0, but the layout is NOT staggered, as is assumed in oi.py line 822. You can tell these probes apart from the older ones because of the imDatPrb_pn, which, for example would be NP1015 for the 1.0 NHP short linear probe.

Can you update io.py to handle these cases?

samuelgarcia, 2023-02-27:
Hi Jonathan.
Would you have a spikeglx meta file to provide us to make the fix ?
Thanks.

JonathanAMichaels, 2023-02-27:
Sure:
https://www.dropbox.com/s/p6fek7l482rfa59/022423_g0_t0.imec0.ap.meta?dl=0

samuelgarcia, 2023-02-27:
Merci.

closed 2023-03-30 by h-mayorquin