SpikeInterface/probeinterface #182: Future plans for probes in probe library?

dbirman, 2023-04-07:
Hi, I'm the developer of [Pinpoint ](https://github.com/VirtualBrainLab/Pinpoint) and I think it would be really nice to use probeinterface to automatically generate 3D models of the many probes people are using for electrophysiology.

I just looked at the probeinterface library and I'm only seeing a handful of probe options. Are there plans to expand the library to other really common probe types such as Neuropixels?

alejoe91, 2023-06-05:
Hi @dbirman 

Sorry for my late reply on this. You are right, currently the Neuropixels probes are created dynamically from the meta/settings files of SpikeGLX/Open Ephys. 

However, I think it makes absolute sense to add the full geometries in the probeinterface library and then use the settings files to dynamically wire the connected channels. 

In this case, loading the probe from the library will return a full probe, while the `read_*` functions will fetch the full probe from the library and slice it according to the channel selection. Do you have the csv files for all NP probes?

dbirman, 2023-06-06:
No worries. I have these files for NP1/NP2 probes, not for any of the more atypical varieties (like ultra,  etc). NP1 is above, here's the NP2 file: [neuropixels_2.csv](https://github.com/SpikeInterface/probeinterface/files/11658797/neuropixels_2.csv). This defines the channel geometry along a single probe shank relative to the tip, so you have to extrapolate to get 4 shanks.

samuelgarcia, 2023-06-06:
Maybe this files could be converted to the probeinterface json format et push to the libray with a name like "NP1-all-channels".
What do you think ?

danielpollak, 2024-03-27:
Not sure if this hasn't already been done somewhere, but I thought it might be useful to share this quick 'n dirty first- or all-bank code snippet for NP1.0:

```

from probeinterface.plotting import plot_probe
import probeinterface
total=384 #, 960
NP1 = probeinterface.generate_multi_columns_probe(
    4, num_contact_per_column=total//4, xpitch=16, ypitch=40,
    y_shift_per_column=[0,20,0,20], contact_shapes="square", contact_shape_params={"width":12},)

plot_probe(NP1, with_contact_id=True, title="NP1.0")
```

WeissShahaf, 2024-05-02:
was this implemented?
@samuelgarcia a function to export from probeinteface to pinpoint would be greatly appriciated

@dbirman an option to upload a probeinteface  probe design into the Pinpoint GUI would also be great. or even just a convertion script / function

dbirman, 2024-05-22:
I didn't end up working more on this, although I do want to find a solution for easier importing of probes into Pinpoint. The probeinterface_library repo doesn't seem to be getting any updates, so I assume when people use probeinterface they are using it directly without loading the specs for a probe from somewhere?

I could use probeinterface to define the metadata for probes and then build them into Pinpoint, which would be a little easier for people to make contributions, just not sure if this is the right tool for that!

WeissShahaf, 2024-05-23:
@db or each user makes his own probe that he saves locally.

But I guess you're right. Picking one platform isn't the way.
Perhaps you can just publish the Metadata pinpoint needs. And have an import function that looks for that. 
Then it's up to the user to create that data. However they choose to.

alejoe91, 2025-05-02:
We added all neuropixels probes to the library recently: https://github.com/SpikeInterface/probeinterface_library/tree/main/neuropixels

Closing this

closed 2025-05-02 by alejoe91

dbirman, 2025-05-02:
Nice, finally time to revisit this and get it working with Pinpoint!