SpikeInterface/probeinterface #185: Wrong probe map from open ephys and unexpected position from peak location

Kayv-cmb, 2023-04-17:
Hi all,

Here is my issue (with some plot). Also open in spikeinterface https://github.com/SpikeInterface/spikeinterface/issues/1546#issue-1671534965 
I started using a 4 shanks neuropixel probe for my recording and when I try to load the open ephys (version 0.64)  recording into spikeinterface (version 0.96.1), my channel location are different from the one I have in my settings.xml
```
from probeinterface import Probe, ProbeGroup
from probeinterface import generate_linear_probe, generate_multi_shank

directory = '/camp/home/combadk/working/raw_data/raw_NP/NP_230315_A/2023-03-15_18-07-02'
raw_rec = si.read_openephys(directory, stream_id ='0',block_index=1)
raw_rec

fig, ax = plt.subplots(figsize=(15, 10))
si.plot_probe_map(raw_rec, ax=ax, with_channel_ids=True)
ax.set_ylim(-800, 750)
```

It give this result
[image]

When I would expect this probe map, that I created from the settings.xml from open ephys.
[image]

I changed the probe with those line and did some preprocessing.
```
df_probe = raw_rec.get_probe().to_dataframe()
chan_map = io.loadmat( '/camp/home/combadk/working/combadk/Kilosort/configFiles/chanMap.mat')
chan_location = pd.DataFrame(chan_map['xcoords'][0,:],chan_map['ycoords'][0,:])
df_probe['x'] = chan_map['xcoords'][0,:]
df_probe['y'] = chan_map['ycoords'][0,:]
df_probe

probe = Probe.from_dataframe(df_probe)
probe.set_device_channel_indices(df_probe.index)
rec = raw_rec.set_probe(probe)
rec1 = si.highpass_filter(rec, freq_min=400.)
rec4 = si.common_reference(rec1, operator="median", reference="global")
preprocess_recording = rec4
job_kwargs = dict(n_jobs=40, chunk_duration='1s', progress_bar=True)
peaks = detect_peaks(preprocess_recording,  method='locally_exclusive', noise_levels=noise_levels_int16,
                     detect_threshold=5, local_radius_um=50., **job_kwargs)

peak_locations = localize_peaks(preprocess_recording, peaks, method='center_of_mass', **job_kwargs)
# check for drifts
fs = rec.sampling_frequency
fig, ax = plt.subplots(figsize=(10, 8))
ax.scatter(peaks['sample_ind'] / fs, peak_locations['y'], color='k', marker='.',  alpha=0.002)
# we can also use the peak location estimates to have an insight of cluster separation before sorting
fig, ax = plt.subplots(figsize=(15, 10))
si.plot_probe_map(preprocess_recording, ax=ax, with_channel_ids=True)
ax.set_ylim(-800, 800)

ax.scatter(peak_locations['x'], peak_locations['y'], color='purple', alpha=0.002)
```

However the result from me checking on the drift looks like my first probe map which I expect to be wrong.
[image]

Any help would be appreciated! Thank you in advance!

alejoe91, 2023-04-19:
@Kayv-cmb  Could you send us the `settings.xml` file? It would help us find and fix the bug!

Kayv-cmb, 2023-04-19:
[settings_2.xml.zip](https://github.com/SpikeInterface/probeinterface/files/11273801/settings_2.xml.zip)
Here a compressed version of my xml file(compressed just to upload on github)
Also I upgraded my version of spikeinterface to the latest one 0.97.1 and it seems to have solve the problem. I think the issue can be close.

closed 2023-04-19 by alejoe91

Kayv-cmb, 2023-04-19:
I think it should be ok if you use it for internal testing.

alejoe91, 2023-04-19:
If it's ok, we would like to include it in the set of test files that we test against on GitHub: https://github.com/SpikeInterface/probeinterface/tree/main/tests/data/openephys

It's only the `settings.xml` file, so there is no ephys data associated to it. Let me knoe

Kayv-cmb, 2023-04-20:
Hi,
Feel free to use it.