SpikeInterface/probeinterface #186: IMEC probes

takemarull, 2023-04-19:
Hi all,

As others have posted, I'm currently using probes not included in the probeinterface library (specifically the IMEC neuropixel 1.0 probe with 384 channels). Would it be possible to add this to the list? Thank you!

alejoe91, 2023-04-19:
I guess you're recording with SpikeGLX or Open Ephys? You can use the `read_spikeglx(path-to-meta-file)` and `read_openephys(path-to-settings-file)`

takemarull, 2023-04-19:
Thank you! Yes, we're recording with SpikeGLX--after I use read_spikeglx, how do I add the resulting SpikeGLXRecordingExtractor to the BinaryRecordingExtractor?

alejoe91, 2023-04-19:
You should just use the read SpikeGLX function of spikeinterface! That automatically loads the probe. See this tutorial: https://spikeinterface.readthedocs.io/en/latest/how_to/analyse_neuropixels.html

closed 2023-04-19 by takemarull