SpikeInterface/probeinterface #189: Parse spikeGLX meta files that have been processed by catGT

emilyasterjones, 2023-06-01:
CatGT adds its command string to the .ap.meta and .lf.meta files when it's run. The command string contains several '=', which breaks [this line](https://github.com/SpikeInterface/probeinterface/blob/96e67da56286e0c162324d129a20e50a2cf39658/probeinterface/io.py#LL1168C13-L1168C13) in parse_spikeglx_meta (`key, val = line.split('=')`). The code is expecting to unpack 2 values from line.split, but receives many more than that.

This could be fixed by adding:

```
if 'catGTCmdline0' in line:
            continue
```

before this line.

alejoe91, 2023-06-02:
Thanks @emilyasterjones 

Could you share a meta file modified by CatGT so we can add some additional tests?

brandonStell, 2023-06-06:
We're seeing the same issue.  Here is a .meta (the third line seems to be the problem):

```
acqApLfSy=384,384,1
appVersion=20230120
catGTCmdline0=<CatGT -dir=/media/setups/bsinvivo3/neuropixels/2023_04_27 -run=77.230125_2023-04-27 -g=0 -t=0 -prb_fld -ni -ap -lf -prb=0 -gblcar -dest=/media/bs/tmp_working/ecephys -out_prb_fld>
catGVals=0,0
catNFiles=1
catTVals=0,0
fileCreateTime=2023-06-05T17:20:39
fileCreateTime_original=2023-04-27T15:21:38
fileName=/media/bs/tmp_working/ecephys/catgt_77.230125_2023-04-27_g0/77.230125_2023-04-27_g0_imec0/77.230125_2023-04-27_g0_tcat.imec0.ap.bin
fileSHA1=0
fileSizeBytes=98624725430
fileTimeSecs=4269.447345892785
firstSample=48994605
gateMode=Immediate
imAiRangeMax=0.6
imAiRangeMin=-0.6
imCalibrated=true
imDatApi=3.57
imDatBs_fw=2.0.169
imDatBsc_fw=3.2.176
imDatBsc_hw=2.2
imDatBsc_pn=NP2_QBSC_00
imDatBsc_sn=22030965
imDatFx_hw=4.13
imDatFx_pn=NP2_FLEX_0
imDatFx_sn=0
imDatHs_hw=5.1
imDatHs_pn=NP2_HS_30
imDatHs_sn=852
imDatPrb_dock=1
imDatPrb_pn=PRB_1_4_0480_1_C
imDatPrb_port=1
imDatPrb_slot=4
imDatPrb_sn=19454409771
imDatPrb_type=0
imIsSvyRun=false
imLEDEnable=false
imMaxInt=512
imSampRate=30000.149579831934
imStdby=
imSvyMaxBnk=-1
imSvySecPerBnk=35
imTrgRising=true
imTrgSource=0
imroFile=C:/Users/bs_spikeglx/Desktop/trst.imro
nDataDirs=1
nSavedChans=385
snsApLfSy=384,0,1
snsSaveChanSubset=0:383,768
syncImInputSlot=4
syncSourceIdx=3
syncSourcePeriod=1
trigMode=Immediate
typeImEnabled=1
typeNiEnabled=1
typeObEnabled=0
typeThis=imec
userNotes=
~imroTbl=(0,384)(0 0 1 500 50 1)(1 0 1 500 50 1)(2 0 1 500 50 1)(3 0 1 500 50 1)(4 0 1 500 50 1)(5 0 1 500 50 1)(6 0 1 500 50 1)(7 0 1 500 50 1)(8 0 1 500 50 1)(9 0 1 500 50 1)(10 0 1 500 50 1)(11 0 1 500 50 1)(12 0 1 500 50 1)(13 0 1 500 50 1)(14 0 1 500 50 1)(15 0 1 500 50 1)(16 0 1 500 50 1)(17 0 1 500 50 1)(18 0 1 500 50 1)(19 0 1 500 50 1)(20 0 1 500 50 1)(21 0 1 500 50 1)(22 0 1 500 50 1)(23 0 1 500 50 1)(24 0 1 500 50 1)(25 0 1 500 50 1)(26 0 1 500 50 1)(27 0 1 500 50 1)(28 0 1 500 50 1)(29 0 1 500 50 1)(30 0 1 500 50 1)(31 0 1 500 50 1)(32 0 1 500 50 1)(33 0 1 500 50 1)(34 0 1 500 50 1)(35 0 1 500 50 1)(36 0 1 500 50 1)(37 0 1 500 50 1)(38 0 1 500 50 1)(39 0 1 500 50 1)(40 0 1 500 50 1)(41 0 1 500 50 1)(42 0 1 500 50 1)(43 0 1 500 50 1)(44 0 1 500 50 1)(45 0 1 500 50 1)(46 0 1 500 50 1)(47 0 1 500 50 1)(48 0 1 500 50 1)(49 0 1 500 50 1)(50 0 1 500 50 1)(51 0 1 500 50 1)(52 0 1 500 50 1)(53 0 1 500 50 1)(54 0 1 500 50 1)(55 0 1 500 50 1)(56 0 1 500 50 1)(57 0 1 500 50 1)(58 0 1 500 50 1)(59 0 1 500 50 1)(60 0 1 500 50 1)(61 0 1 500 50 1)(62 0 1 500 50 1)(63 0 1 500 50 1)(64 0 1 500 50 1)(65 0 1 500 50 1)(66 0 1 500 50 1)(67 0 1 500 50 1)(68 0 1 500 50 1)(69 0 1 500 50 1)(70 0 1 500 50 1)(71 0 1 500 50 1)(72 0 1 500 50 1)(73 0 1 500 50 1)(74 0 1 500 50 1)(75 0 1 500 50 1)(76 0 1 500 50 1)(77 0 1 500 50 1)(78 0 1 500 50 1)(79 0 1 500 50 1)(80 0 1 500 50 1)(81 0 1 500 50 1)(82 0 1 500 50 1)(83 0 1 500 50 1)(84 0 1 500 50 1)(85 0 1 500 50 1)(86 0 1 500 50 1)(87 0 1 500 50 1)(88 0 1 500 50 1)(89 0 1 500 50 1)(90 0 1 500 50 1)(91 0 1 500 50 1)(92 0 1 500 50 1)(93 0 1 500 50 1)(94 0 1 500 50 1)(95 0 1 500 50 1)(96 0 1 500 50 1)(97 0 1 500 50 1)(98 0 1 500 50 1)(99 0 1 500 50 1)(100 0 1 500 50 1)(101 0 1 500 50 1)(102 0 1 500 50 1)(103 0 1 500 50 1)(104 0 1 500 50 1)(105 0 1 500 50 1)(106 0 1 500 50 1)(107 0 1 500 50 1)(108 0 1 500 50 1)(109 0 1 500 50 1)(110 0 1 500 50 1)(111 0 1 500 50 1)(112 0 1 500 50 1)(113 0 1 500 50 1)(114 0 1 500 50 1)(115 0 1 500 50 1)(116 0 1 500 50 1)(117 0 1 500 50 1)(118 0 1 500 50 1)(119 0 1 500 50 1)(120 0 1 500 50 1)(121 0 1 500 50 1)(122 0 1 500 50 1)(123 0 1 500 50 1)(124 0 1 500 50 1)(125 0 1 500 50 1)(126 0 1 500 50 1)(127 0 1 500 50 1)(128 0 1 500 50 1)(129 0 1 500 50 1)(130 0 1 500 50 1)(131 0 1 500 50 1)(132 0 1 500 50 1)(133 0 1 500 50 1)(134 0 1 500 50 1)(135 0 1 500 50 1)(136 0 1 500 50 1)(137 0 1 500 50 1)(138 0 1 500 50 1)(139 0 1 500 50 1)(140 0 1 500 50 1)(141 0 1 500 50 1)(142 0 1 500 50 1)(143 0 1 500 50 1)(144 0 1 500 50 1)(145 0 1 500 50 1)(146 0 1 500 50 1)(147 0 1 500 50 1)(148 0 1 500 50 1)(149 0 1 500 50 1)(150 0 1 500 50 1)(151 0 1 500 50 1)(152 0 1 500 50 1)(153 0 1 500 50 1)(154 0 1 500 50 1)(155 0 1 500 50 1)(156 0 1 500 50 1)(157 0 1 500 50 1)(158 0 1 500 50 1)(159 0 1 500 50 1)(160 0 1 500 50 1)(161 0 1 500 50 1)(162 0 1 500 50 1)(163 0 1 500 50 1)(164 0 1 500 50 1)(165 0 1 500 50 1)(166 0 1 500 50 1)(167 0 1 500 50 1)(168 0 1 500 50 1)(169 0 1 500 50 1)(170 0 1 500 50 1)(171 0 1 500 50 1)(172 0 1 500 50 1)(173 0 1 500 50 1)(174 0 1 500 50 1)(175 0 1 500 50 1)(176 0 1 500 50 1)(177 0 1 500 50 1)(178 0 1 500 50 1)(179 0 1 500 50 1)(180 0 1 500 50 1)(181 0 1 500 50 1)(182 0 1 500 50 1)(183 0 1 500 50 1)(184 0 1 500 50 1)(185 0 1 500 50 1)(186 0 1 500 50 1)(187 0 1 500 50 1)(188 0 1 500 50 1)(189 0 1 500 50 1)(190 0 1 500 50 1)(191 0 1 500 50 1)(192 0 1 500 50 1)(193 0 1 500 50 1)(194 0 1 500 50 1)(195 0 1 500 50 1)(196 0 1 500 50 1)(197 0 1 500 50 1)(198 0 1 500 50 1)(199 0 1 500 50 1)(200 0 1 500 50 1)(201 0 1 500 50 1)(202 0 1 500 50 1)(203 0 1 500 50 1)(204 0 1 500 50 1)(205 0 1 500 50 1)(206 0 1 500 50 1)(207 0 1 500 50 1)(208 0 1 500 50 1)(209 0 1 500 50 1)(210 0 1 500 50 1)(211 0 1 500 50 1)(212 0 1 500 50 1)(213 0 1 500 50 1)(214 0 1 500 50 1)(215 0 1 500 50 1)(216 0 1 500 50 1)(217 0 1 500 50 1)(218 0 1 500 50 1)(219 0 1 500 50 1)(220 0 1 500 50 1)(221 0 1 500 50 1)(222 0 1 500 50 1)(223 0 1 500 50 1)(224 0 1 500 50 1)(225 0 1 500 50 1)(226 0 1 500 50 1)(227 0 1 5 [... cut ...] )(0:1:169:1)(0:0:170:1)(0:1:170:1)(0:0:171:1)(0:1:171:1)(0:0:172:1)(0:1:172:1)(0:0:173:1)(0:1:173:1)(0:0:174:1)(0:1:174:1)(0:0:175:1)(0:1:175:1)(0:0:176:1)(0:1:176:1)(0:0:177:1)(0:1:177:1)(0:0:178:1)(0:1:178:1)(0:0:179:1)(0:1:179:1)(0:0:180:1)(0:1:180:1)(0:0:181:1)(0:1:181:1)(0:0:182:1)(0:1:182:1)(0:0:183:1)(0:1:183:1)(0:0:184:1)(0:1:184:1)(0:0:185:1)(0:1:185:1)(0:0:186:1)(0:1:186:1)(0:0:187:1)(0:1:187:1)(0:0:188:1)(0:1:188:1)(0:0:189:1)(0:1:189:1)(0:0:190:1)(0:1:190:1)(0:0:191:1)(0:1:191:1)
```

alejoe91, 2023-06-06:
@brandonStell @emilyasterjones 

Should be fixed here: #193

@brandonStell I added the file you provided in the test files. Is that ok?

PR #193 (2023-06-06): Fix CatGT parsing
Fixes #189