SpikeInterface/probeinterface #194: read_probeinterface() does not read in device_channel_indices

ThetaS, 2023-06-08:
Hello,

I got this problem when attempting to attach a probe read in using read_probeinterface(). Since I have only one probe, I first convert the read in probegroup to dataframe then instantiate a Probe object from dataframe. The .json file was able to retain the device indices I saved but the read in function didn't take this column in as it was not in the dataframe. Is there something I misunderstood and did wrong, or it was a glitch of the function?

Thanks,

samuelgarcia, 2023-06-08:
Hi @ThetaS.
Can you share the piece of code to generate the probe + the dataframe + the json file ?

ThetaS, 2023-06-12:
Thanks for the response and sorry for the delay! Sure, here are the code snippets:
[image]

This is the calling from my analysis code:
[image]

I'm working with the UCLA Masmanidis probes and had quite a hard to generate the probe map because they are checkerboard patterned and multi-shank. Hope I didn't miss any shortcuts and solutions for handling this problem.

Thank you very much.

ThetaS

[128ANbottom_map.zip](https://github.com/SpikeInterface/probeinterface/files/11718014/128ANbottom_map.zip)

ThetaS, 2023-06-12:
[128_ANbottom_map.csv](https://github.com/SpikeInterface/probeinterface/files/11718030/128_ANbottom_map.csv)
This is the CSV associated. Basically, the information from the Masmanidis lab's Matlab program and the wiring configuration to Intan 128 headstage.

Best

samuelgarcia, 2023-06-12:
Hi,
You have device_channel_ids, in probeinterface it should device_channel_indices and it should be starting from 0.
So a zero-based index and not an "id" (identifier).
contact_id = id = str or int = zeros-based orone-based
device_channel_indices = index = zero-based in the python ecosystem

samuelgarcia, 2023-06-16:
The plan axes is a bit more complicated. This is used internally for rotation. And so not so well documentated.

On this example : https://probeinterface.readthedocs.io/en/main/examples/ex_08_more_plotting_examples.html#sphx-glr-examples-ex-08-more-plotting-examples-py
you can see that some some probe are rotated. In that case you should inspect the plane_axes and you will understand how this is used.

By default, you should not use this plan_axes and everything should be OK and with no rotation.

ThetaS, 2023-06-17:
Yes, I tried deleting the plane_axes parameter and the plotting works. But the previous question, I tried the keyword "device_channel_indices" and zero-based my indices and it failed to work. 

[image]
[image]
[image]

The save part is the same as above, key modifications are shown in the images.

alejoe91, 2023-08-23:
I think the problem is how you're setting the probe in SpikeInterface!

You need to use `in_place=True` if you want to modify an existing object. Otherwise the `set_probe/probegroup` will return a separate object (see here: https://spikeinterface.readthedocs.io/en/latest/modules/core.html#handling-probes)

Basically you can do:
```
data.set_probegroup(probe, group_mode="by_shank", in_place=True)
# or
daa_w_probe = data.set_probegroup(probe, group_mode="by_shank") 
# then continue with data_w_probe
```

closed 2025-05-02 by alejoe91