SpikeInterface/probeinterface #200: Multi Shank configuration in Neuropixels 2 does not result in correct channel layout

manimoh, 2023-07-20:
Hi,

I recorded a session using a 4 shank NeuroPixels 2.0 using SpikeGLX acquisition software, where I recorded 384 channels split equally between middle two shanks. However when I try to plot the prob using the following code, the result puts all the contacts on the the first shank:

```
import spikeinterface.full as si
import numpy as np
import matplotlib.pyplot as plt
import probeinterface as pif
import scipy.io as scio
import os

meta_filename = os.path.join('path to meta_file', 'npx2.imec0.ap.meta')
probe = pif.read_spikeglx(meta_filename)
pif.plotting.plot_probe(probe)
```

[image]

I had tried reading a raw recording, and then get the probe using `get_probe()`, but that too yields the same result as above. But since the title says 2 shanks, I dug a little deeper to see what exactly is happening, and saw that probe.contact_positions are not set according to the actual configuration and the output of `probe._contact_ids` seems to have the correct shank and electrode information:

```
array(['s1e0', 's1e1', 's1e2', 's1e3', 's1e4', 's1e5', 's1e6', 's1e7',
       's1e8', 's1e9', 's1e10', 's1e11', 's1e12', 's1e13', 's1e14',
       's1e15', 's2e16', 's2e17', 's2e18', 's2e19', 's2e20', 's2e21',
       's2e22', 's2e23', 's2e24', 's2e25', 's2e26', 's2e27', 's2e28',
       's2e29', 's2e30', 's2e31', 's2e32', 's2e33', 's2e34', 's2e35',
       's2e36', 's2e37', 's2e38', 's2e39', 's2e40', 's2e41', 's2e42',
       's2e43', 's2e44', 's2e45', 's2e46', 's2e47', 's2e48', 's2e49',
       's2e50', 's2e51', 's2e52', 's2e53', 's2e54', 's2e55', 's2e56',
       's2e57', 's2e58', 's2e59', 's2e60', 's2e61', 's2e62', 's2e63',
       's2e64', 's2e65', 's2e66', 's2e67', 's2e68', 's2e69', 's2e70',
       's2e71', 's2e72', 's2e73', 's2e74', 's2e75', 's2e76', 's2e77',
       's2e78', 's2e79', 's2e80', 's2e81', 's2e82', 's2e83', 's2e84',
       's2e85', 's2e86', 's2e87', 's2e88', 's2e89', 's2e90', 's2e91',
       's2e92', 's2e93', 's2e94', 's2e95', 's2e96', 's2e97', 's2e98',
       's2e99', 's2e100', 's2e101', 's2e102', 's2e103', 's2e104',
       's2e105', 's2e106', 's2e107', 's2e108', 's2e109', 's2e110',
       's2e111', 's2e112', 's2e113', 's2e114', 's2e115', 's2e116',
       's2e117', 's2e118', 's2e119', 's2e120', 's2e121', 's2e122',
       's2e123', 's2e124', 's2e125', 's2e126', 's2e127', 's2e128',
       's2e129', 's2e130', 's2e131', 's2e132', 's2e133', 's2e134',
       's2e135', 's2e136', 's2e137', 's2e138', 's2e139', 's2e140',
       's2e141', 's2e142', 's2e143', 's2e144', 's2e145', 's2e146',
       's2e147', 's2e148', 's2e149', 's2e150', 's2e151', 's2e152',
       's2e153', 's2e154', 's2e155', 's2e156', 's2e157', 's2e158',
       's2e159', 's2e160', 's2e161', 's2e162', 's2e163', 's2e164',
       's2e165', 's2e166', 's2e167', 's2e168', 's2e169', 's2e170',
       's2e171', 's2e172', 's2e173', 's2e174', 's2e175', 's2e176',
       's2e177', 's2e178', 's2e179', 's2e180', 's2e181', 's2e182',
       's2e183', 's2e184', 's2e185', 's2e186', 's2e187', 's2e188',
       's2e189', 's2e190', 's2e191', 's1e192', 's1e193', 's1e194',
       's1e195', 's1e196', 's1e197', 's1e198', 's1e199', 's1e200',
       's1e201', 's1e202', 's1e203', 's1e204', 's1e205', 's1e206',
       's1e207', 's1e208', 's1e209', 's1e210', 's1e211', 's1e212',
       's1e213', 's1e214', 's1e215', 's1e216', 's1e217', 's1e218',
       's1e219', 's1e220', 's1e221', 's1e222', 's1e223', 's1e224',
       's1e225', 's1e226', 's1e227', 's1e228', 's1e229', 's1e230',
       's1e231', 's1e232', 's1e233', 's1e234', 's1e235', 's1e236',
       's1e237', 's1e238', 's1e239', 's2e240', 's2e241', 's2e242',
       's2e243', 's2e244', 's2e245', 's2e246', 's2e247', 's2e248',
       's2e249', 's2e250', 's2e251', 's2e252', 's2e253', 's2e254',
       's2e255', 's1e256', 's1e257', 's1e258', 's1e259', 's1e260',
       's1e261', 's1e262', 's1e263', 's1e264', 's1e265', 's1e266',
       's1e267', 's1e268', 's1e269', 's1e270', 's1e271', 's1e272',
       's1e273', 's1e274', 's1e275', 's1e276', 's1e277', 's1e278',
       's1e279', 's1e280', 's1e281', 's1e282', 's1e283', 's1e284',
       's1e285', 's1e286', 's1e287', 's1e288', 's1e289', 's1e290',
       's1e291', 's1e292', 's1e293', 's1e294', 's1e295', 's1e296',
       's1e297', 's1e298', 's1e299', 's1e300', 's1e301', 's1e302',
       's1e303', 's1e304', 's1e305', 's1e306', 's1e307', 's1e308',
       's1e309', 's1e310', 's1e311', 's1e312', 's1e313', 's1e314',
       's1e315', 's1e316', 's1e317', 's1e318', 's1e319', 's1e320',
       's1e321', 's1e322', 's1e323', 's1e324', 's1e325', 's1e326',
       's1e327', 's1e328', 's1e329', 's1e330', 's1e331', 's1e332',
       's1e333', 's1e334', 's1e335', 's1e336', 's1e337', 's1e338',
       's1e339', 's1e340', 's1e341', 's1e342', 's1e343', 's1e344',
       's1e345', 's1e346', 's1e347', 's1e348', 's1e349', 's1e350',
       's1e351', 's1e352', 's1e353', 's1e354', 's1e355', 's1e356',
       's1e357', 's1e358', 's1e359', 's1e360', 's1e361', 's1e362',
       's1e363', 's1e364', 's1e365', 's1e366', 's1e367', 's1e368',
       's1e369', 's1e370', 's1e371', 's1e372', 's1e373', 's1e374',
       's1e375', 's1e376', 's1e377', 's1e378', 's1e379', 's1e380',
       's1e381', 's1e382', 's1e383'], dtype='<U6') 
```

And what I found is that in probeinterface/io.py under _read_imro_string() to create the probe, [the lines](https://github.com/SpikeInterface/probeinterface/blob/3fb8864297cdecdcdd97f8d900960e452f27d0c6/src/probeinterface/io.py#L971C1-L979C49) that compute the x and y position don't take into account the shift due to the electrodes being on a different shank, but it does the shanks shift for the `planar contours` to draw the body of the probe

I am attaching the .meta and .imro file so that you can run it on your end and see if it behaves in the same way:

[probeInterfaceIssue.zip](https://github.com/SpikeInterface/probeinterface/files/12113436/probeInterfaceIssue.zip)

samuelgarcia, 2023-07-21:
Hi @manimoh.
Thank you for reporting we will have a look.

alejoe91, 2023-07-31:
@manimoh 

Does this config look correct?

[image]

I made it with this PR, which seems to fix the problem: #205

manimoh, 2023-07-31:
@alejoe91 
Not quite, the final result should be sets of channels on the middle 2 shanks with the same y coordinates (and not staggered as shown above)
But I think these [lines](https://github.com/SpikeInterface/probeinterface/blob/3fb8864297cdecdcdd97f8d900960e452f27d0c6/src/probeinterface/io.py#L965-L966) are causing this bug,  They are looking for key `elecT_ids` (The capitalization is to highlight the problem)  in the dictionary `contact_info` whereas the hard-coded dictionary above uses the key `elec_ids` in these [lines ](https://github.com/SpikeInterface/probeinterface/blob/3fb8864297cdecdcdd97f8d900960e452f27d0c6/src/probeinterface/io.py#L722-L739)and elsewhere in the code too. As a  result, the `elec_ids` are getting calculated according to a [different logic ](https://github.com/SpikeInterface/probeinterface/blob/3fb8864297cdecdcdd97f8d900960e452f27d0c6/src/probeinterface/io.py#L967C1-L969C45)and that is causing the discrepancy in the final position [here](https://github.com/SpikeInterface/probeinterface/blob/3fb8864297cdecdcdd97f8d900960e452f27d0c6/src/probeinterface/io.py#L972C5-L972C66)

alejoe91, 2023-08-02:
Thanks @manimoh 

I changed `elect_ids` to `elec_ids` and now the layout looks as follows:

[image]

I guess that's correct!! Thank you for spotting the error! We'll make a release soon ;)

manimoh, 2023-08-07:
Yeah that's perfect! Closing this :)