SpikeInterface/probeinterface #201: Channel_ids not showing in plot_probe_map widget

SusanL82, 2023-07-24:
Hello, 

I am trying to generate a PRB file for spike sorting with klustakwik for my neuralynx recordings. I am trying to use the  plot_probe_map widget to check if my channel mapping is correct, because the channel order for the NLX format is not very straightforward.
For example, my first tetrode has channel_ids 12-15, which I have set to channel group 0. If I have understood the probeinterface documentation correctly, the correct channel indices for these channels should be 0, 11, 22, 33 for the first tetrode (output of np.where(tetgrouping==0) ) . 

However, when I try to plot a probe map, my channel ids are not showing up. Am I defining my channels incorrectly? Or is it because I am using SI 0.95.1? 

Somewhat related: I still need to add some method to exclude bad/broken channels. Is this best done by making a probe with fewer contacts, or should I rather find a way to do this in within spikeinterface instead? 

 
My probe-related code looks as follows so far

`import spikeinterface.widgets as swi
import spikeinterface.extractors as se
import numpy as np
from probeinterface import ProbeGroup, generate_tetrode

InFolder = "D:/HD180/2022-09-27_16-10-48"

folder_path_ntt = InFolder

# Read NCS
recording_ncs = se.read_neuralynx(folder_path_ntt)

# output from recording_ncs.get_channel_ids()
#        array(['12', '29', '30', '31', '24', '25', '26', '27', '20', '21', '22',
#        '13', '23', '16', '17', '18', '19', '4', '5', '6', '7', '0', '14',
#        '1', '2', '3', '48', '49', '50', '51', '52', '53', '54', '15',
#        '55', '32', '33', '34', '35', '36', '37', '38', '39', '40', '8',
#        '41', '42', '43', '44', '45', '46', '47', '56', '57', '58', '9',
#        '59', '60', '61', '62', '63', '10', '11', '28'], dtype='<U64')

#add tetrode number for each channel
tetgrouping = np.array([0, 2, 2, 2, 3, 3, 3, 3, 4, 4, 4,
               0, 4, 5, 5, 5, 5, 6, 6, 6, 6, 7, 0,
               7, 7, 7, 8, 8, 8, 8, 9, 9, 9, 0,
               9, 10, 10, 10, 10, 11, 11, 11, 11, 12, 1,
               12, 12, 12, 13, 13, 13, 13, 14, 14, 14, 1,
               14, 15, 15, 15, 15, 1, 1, 2])

recording_ncs.set_channel_groups(channel_ids = recording_ncs.get_channel_ids(),groups = tetgrouping)

# make probe with 16 tetrodes
probegroup = ProbeGroup()
for i in range(16):
    tetrode = generate_tetrode()
    tetrode.move([i * 50, 0])
    tetrode.set_device_channel_indices(np.where(tetgrouping==i))
    probegroup.add_probe(tetrode)

rec_withprobe = recording_ncs.set_probegroup(probegroup)

swi.plot_probe_map(rec_withprobe, with_channel_ids=True)

# save as PRB? Maybe just sort with klustakwik directly?`

And the probe map looks like this:
[image]

SusanL82, 2023-07-27:
I figured out the problem with plotting: 
I forgot to add the `group_mode='by_probe'` setting in `rec_withprobe = recording_ncs.set_probegroup(probegroup)`
Everything now shows up fine (the text is just a little large and overlaps, but that's not a big problem).

samuelgarcia, 2023-07-28:
Glad you find the solution.

About the PRB, if you use `run_sorter('klustawik', recording ...)` you do have to worry about saving the prb file this is done automatically.

About klustakwik, this sorter is very old not very accurate, not maintain and need python <3.7
In short this is not a good choice! Maybe you test sorter like montainsort5 or tridesclous.

alejoe91, 2023-07-31:
@SusanL82 closing then!

closed 2023-07-31 by alejoe91