SpikeInterface/probeinterface #202: Add list of common adapter connectors and headstages available to probe.wiring_to_device in the documentation

vncntprvst, 2023-07-26:
Hi,
I'd like to use `probe.wiring_to_device` to map the channel indices. The [documentation](https://probeinterface.readthedocs.io/en/main/examples/ex_11_automatic_wiring.html) does not list what adapters and headstages are available. Would it be possible to add that list to the documentation? 
Thanks

alejoe91, 2023-07-27:
Thanks @vncntprvst for the suggestion! 

We'll definitely do!

samuelgarcia, 2023-07-27:
We already have this: 
https://github.com/SpikeInterface/probeinterface/blob/main/src/probeinterface/wiring.py#L53C5-L53C29

vncntprvst, 2023-07-27:
> not in the docs ;)

Samuel is a code-base clairvoyant. He can read the docs directly from the source. :)

alejoe91, 2023-08-23:
@vncntprvst would something like #213 work?

vncntprvst, 2023-08-25:
Yes, definitely, that would work. Thanks

closed 2023-09-12 by alejoe91