SpikeInterface/probeinterface #208: NeuroNexus one-based indexing of probe

Sietse-k, 2023-08-15:
Hi, 

In the 'automatic wiring tutorial' (https://probeinterface.readthedocs.io/en/main/examples/ex_11_automatic_wiring.html), there is an example wiring of a NeuroNexus electrode ( 'A1x32-Poly3-10mm-50-177' ). 

[image]

At the bottom of the tutorial it is stated that:

 "In this figure we have 2 numbers for each contact:
the upper number “prbXX” is the Neuronexus index (one-based)
the lower “devXX” is the channel on the Intan device (zero-based)

I am wondering how it is possible to start the prbXX indexing at 1 instead of 0? Since this "prbXX" index is automatically generated based on X and Y coordinates (at least when I try to replicate this exact probe). 
[image]

Any insights would be appreciated, 

thanks, 

Sietse

Sietse-k, 2023-08-15:
Found the solution. You can set it manually using: 

probe.annotations['first_index'] = 1
probe.annotations['name'] = 'A1x32-Poly3-10mm-50-177'
probe.annotations['manufacturer'] = 'neuronexus'

for example, 

Cheers

alejoe91, 2023-08-23:
So in this case the contact id (i.e. the `prbXXX`) is not automatically generated, but it corresponds to the catalog of NeuroNexus (in this case). In many of these probe catalogs, the contact ids start from "1", but the device channel index has to start from 0, since it corresponds to the index of the trace of that channel (0-based in Python). Hope this makes it clearer! Closing

closed 2023-08-23 by alejoe91