SpikeInterface/probeinterface #210: Issues loading probes macos

mzabolocki, 2023-08-18:
Hi, I am having the same issue as previous (https://github.com/SpikeInterface/probeinterface/issues/95) with macos. I am also trying to import manufacturer Cambridge Neurotech and probe name ASSY-77-P-2.

Version for probe interface is: 0.2.17

Any help would be appreciated!

alejoe91, 2023-08-23:
Hi @mzabolocki 

You have access to an internet connection when requesting the probe, right? :) 

In that case, not really sure what's going on. @vincisLab did you have to do something in particular to fix #95?

mzabolocki, 2023-08-23:
Yes. Nothing is blocking the connection either. 

To be more specific. The issue lies with the Cambridge neurotech probe listed above. I tested the probe 'ASSY-37-E-1' listed on the SI documentation (https://spikeinterface.readthedocs.io/en/latest/modules_gallery/core/plot_3_handle_probe_info.html#sphx-glr-modules-gallery-core-plot-3-handle-probe-info-py) and had no issues. This seems to be a previous issue also (#95). Thanks again!

alejoe91, 2023-08-23:
Thanks for the clarification. I can download both probes on my side..can you try to remove the folder where probeinterface caches the downloded probes?

It should be `$HOME/.config/probeinterface/library`. Then try to download both again! It could be that the 'ASSY-37-E-1' is reloaded from the cache, and not downloaded again...

mzabolocki, 2023-12-04:
Sorry for the delayed reply! This seems to be working fine now. Seemed to be some admin / firewall rights at the Institute I relocated to.

closed 2023-12-04 by mzabolocki