SpikeInterface/probeinterface #215: Question about set_device_channel_indices

SupremeLiam, 2023-09-06:
I am using `spikeinterface.extractors.read_alphaomega` for reading AlphaOmega .mpx recording files. 

This file is recorded with Plexon S-Probe which is a 16*1 linear array.

I use `generate_linear_probe` to generate a linear probe, and I want to `set_device_channel_indices`.

However, it seems that the recording channels could not be modified and it reports the following output.

I would like to know how to solve it. Thanks!

```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Cell In[11], line 3
      1 device_channel_indices = [10256, 10257, 10258, 10259, 10260, 10261, 10262, 10263, 10271, 10270, 10269, 10268, 10267, 10266, 10265, 10264]
      2 linear_probe.set_device_channel_indices(device_channel_indices)
----> 3 recording.set_probe(linear_probe)

File ~\.conda\envs\si_env\lib\site-packages\spikeinterface\core\baserecordingsnippets.py:80, in BaseRecordingSnippets.set_probe(self, probe, group_mode, in_place)
     78 probegroup = ProbeGroup()
     79 probegroup.add_probe(probe)
---> 80 return self.set_probes(probegroup, group_mode=group_mode, in_place=in_place)

File ~\.conda\envs\si_env\lib\site-packages\spikeinterface\core\baserecordingsnippets.py:154, in BaseRecordingSnippets.set_probes(self, probe_or_probegroup, group_mode, in_place)
    147 if number_of_device_channel_indices >= self.get_num_channels():
    148     error_msg = (
    149         f"The given Probe have 'device_channel_indices' that do not match channel count \n"
    150         f"{number_of_device_channel_indices} vs {self.get_num_channels()} \n"
    151         f"device_channel_indices are the following: {device_channel_indices} \n"
    152         f"recording channels are the following: {self.get_channel_ids()} \n"
    153     )
--> 154     raise ValueError(error_msg)
    156 new_channel_ids = self.get_channel_ids()[device_channel_indices]
    157 probe_as_numpy_array = probe_as_numpy_array[order]

ValueError: The given Probe have 'device_channel_indices' that do not match channel count 
10271 vs 16 
device_channel_indices are the following: [10256 10257 10258 10259 10260 10261 10262 10263 10264 10265 10266 10267
 10268 10269 10270 10271] 
recording channels are the following: ['10256' '10257' '10258' '10259' '10260' '10261' '10262' '10263' '10264'
 '10265' '10266' '10267' '10268' '10269' '10270' '10271'] 
```

alejoe91, 2023-09-06:
Hi, it seems that your recording has 10271 channels! I guess that's wrong...right? Then the problem is in the alphaomega extractor

SupremeLiam, 2023-09-06:
In fact it only has 16 channels. The “10xxx” indexes are defined by the AlphaOmega SnR recording software, so the recording channels are 10256-10271. I am wondering if there is a way to map the device channels or if it is possible to modify the indexes of recording channels.

SupremeLiam, 2023-09-06:
Although I set the device_channel_indices from 1-16, it still outputs this.
'''---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Cell In[13], line 4
      2 device_channel_indices = [1, 2, 3, 4, 5, 6, 7, 8, 16, 15, 14, 13, 12, 11, 10, 9]
      3 linear_probe.set_device_channel_indices(device_channel_indices)
----> 4 recording.set_probe(linear_probe)

File ~\.conda\envs\si_env\lib\site-packages\spikeinterface\core\baserecordingsnippets.py:80, in BaseRecordingSnippets.set_probe(self, probe, group_mode, in_place)
     78 probegroup = ProbeGroup()
     79 probegroup.add_probe(probe)
---> 80 return self.set_probes(probegroup, group_mode=group_mode, in_place=in_place)

File ~\.conda\envs\si_env\lib\site-packages\spikeinterface\core\baserecordingsnippets.py:154, in BaseRecordingSnippets.set_probes(self, probe_or_probegroup, group_mode, in_place)
    147 if number_of_device_channel_indices >= self.get_num_channels():
    148     error_msg = (
    149         f"The given Probe have 'device_channel_indices' that do not match channel count \n"
    150         f"{number_of_device_channel_indices} vs {self.get_num_channels()} \n"
    151         f"device_channel_indices are the following: {device_channel_indices} \n"
    152         f"recording channels are the following: {self.get_channel_ids()} \n"
    153     )
--> 154     raise ValueError(error_msg)
    156 new_channel_ids = self.get_channel_ids()[device_channel_indices]
    157 probe_as_numpy_array = probe_as_numpy_array[order]

ValueError: The given Probe have 'device_channel_indices' that do not match channel count 
16 vs 16 
device_channel_indices are the following: [ 1  2  3  4  5  6  7  8  9 10 11 12 13 14 15 16] 
recording channels are the following: ['10256' '10257' '10258' '10259' '10260' '10261' '10262' '10263' '10264'
 '10265' '10266' '10267' '10268' '10269' '10270' '10271'] 
'''

samuelgarcia, 2023-09-06:
Hi device_channel_indices is zero based.
Please have a look to this : 
https://probeinterface.readthedocs.io/en/main/examples/ex_05_device_channel_indices.html

closed 2023-09-06 by SupremeLiam

samuelgarcia, 2023-09-06:
No worry.