SpikeInterface/probeinterface #238: Making a probegroup from two of the same cambridge neurotech probes

zm711, 2023-11-03:
Quick question about making probe groups with Cambridge neurotech probes. Cambridge neurotech allows people to buy a probe, eg an H7, or to buy same probe double so two H7s glued together.  Currently I make  probe, wire it and then since the other probe is an exact duplicate I just deep copy that probe. Then I `set_device_indices` as the same as the original probe but offset by the appropriate number of channels (& do the same for the contact numbers. Then I add the original probe and the duplicated (but offset probe) to a probegroup. 

Is there a smarter way to do this?

```python
probe = pi.get_probe(manufacturer = 'cambridgeneurotech', probe_name = 'ASSY-77-H7')
probe.wiring_to_device('ASSY-77>Adpt.A64-Om32_2x-sm-NN>RHD2164')
probe1=deepcopy(probe)
probe1.move([400,0]) # shift so they aren't on top of each other

#offset channel_indices since they are exactly modulo 64
channel_indices = probe.device_channel_indices
probe1.set_device_channel_indices(channel_indices + 64) 

#offset the contact_ids since they are modulo 64
contact_ids = probe.contact_ids
new_contact = []
for idx in range(len(contact_ids)):
    new_contact.append(str(float(contact_ids[idx])+64)) 
new_contacts= np.array(new_contact)
probe1.set_contact_ids(new_contacts)

multishank = pi.ProbeGroup()

multishank.add_probe(probe)
multishank.add_probe(probe1)
```

Then I just sort by `probe_group` in spikeinterface. But if there is an easier way to do this I would be happy to switch this or if I made a wrong step I would also be super happy to be corrected!

alejoe91, 2023-11-03:
Hi @zm711 

- Instead of `deepcopy`, you can use `probe1 = probe.clone()`
- you don't need to reset contact_ids, as we removed the (wrong) requirement that contact ids need to be unique across probes in a probegroup (see [here](https://github.com/SpikeInterface/probeinterface/pull/229/files#diff-ee7ac83f3b80597de69147790858c6c3e73e794677e1fc9e96eebb3acb06b0bdL190-L197))
- for wiring, alternatively you could wire the entire probegroup directly:

```
wiring = pi.wiring.pathways["ASSY-77>Adpt.A64-Om32_2x-sm-NN>RHD2164"]
wiring_group = wiring + list(np.array(wiring) + 64)

multishank = pi.ProbeGroup()
multishank.add_probe(probe)
multishank.add_probe(probe1)

multishank.set_global_device_channel_indices(wiring_group)
```

zm711, 2023-11-03:
Thanks @alejoe91, I will try that as soon my raw file is done reading and close when I test :). Thanks, I figured you guys would have an easier way to do rather than hacking my way through it.

zm711, 2023-11-03:
@alejoe91, the `clone()` is failing?

```ipython
probe1 = probe.clone()
Traceback (most recent call last):

  Cell In[8], line 1
    probe1 = probe.clone()

AttributeError: 'Probe' object has no attribute 'clone'

pi.__version__
Out[9]: '0.2.19'
```

zm711, 2023-11-03:
Is it suppose to be `probe.copy()` instead?

alejoe91, 2023-11-03:
> Is it suppose to be `probe.copy()` instead?

Yes! Sorry about that!

zm711, 2023-11-03:
Cool cool. I'm running the sorting now, but seems to be working and is way simpler. Thanks @alejoe91 :)

closed 2023-11-03 by zm711