SpikeInterface/probeinterface #271: plane_axes ?

borrepp, 2024-05-08:
Hello, 
I'm new to the probeInterface project. I'm working on a sorting pipeline using SpikeInterface from an NWB file. I load the file using spikeInterface and get the positions of the electrodes relative to the probe (x, y, z). But, when I set the contacts to create a the probe, I have the following error:

  File "m:/Monkey_Python/SI2env/S2_sort_testing/SI_tools.py", line 310, in <module>
    locationsProbe = si_recObj.get_property('location')
    probeObj.set_contacts(positions=locationsProbe, shapes='circle', shape_params={'radius':probeInfo['radius']})
  File "m:\Monkey_Python\SI2env\lib\site-packages\probeinterface\probe.py", line 290, in set_contacts
    raise ValueError("For ndim==3, you need to give a 'plane_axes'")
ValueError: For ndim==3, you need to give a 'plane_axes'

I have looked into the Docs, but it is not clear to me what it means the plane_axes variable. Does it have to be always zeros and ones?  What is the meaning of [1 0] or [0 1] on each dimension?

Thanks in advance for your help. 
Best
Pepe

zm711, 2024-05-09:
This is a great question for @samuelgarcia and @alejoe91 I don't use 3d probes much so I can't help with this one. But maybe we need to add some more docs for this!

samuelgarcia, 2024-05-13:
Hi.
This plane_axes is the vector in which the contact shape is defined.

Is is used when we rotate the probe or <when the probe live in a 3D space (but the contact are still 2d shape (circle, rect or square)

See:
  * https://probeinterface.readthedocs.io/en/latest/examples/ex_09_more_complicated_probe.html
  * https://probeinterface.readthedocs.io/en/latest/examples/ex_08_more_plotting_examples.html
  * https://probeinterface.readthedocs.io/en/latest/examples/ex_02_probe_2d_probe_3d.html

borrepp, 2024-05-13:
Thanks, @samuelgarcia  and @zm711

I have seen those examples before, but I will check them again in more detail. 

Just for clarification, the plane_axes defines a vector for a given contact. Based on its shape (2 by nDim), "plane_axes" consists of two coordinates (points) in the Probe coordinate system (2d or 3d). Does it mean that each point defines a unitary vector (starting from the origin) and the plane formed by these two vectors is where the contact is drawn? 

Hope it makes sense the question. 

Thanks again for your help. 

Best
Pepe

samuelgarcia, 2024-05-14:
>  Does it mean that each point defines a unitary vector (starting from the origin) and the plane formed by these two vectors is where the contact is drawn?

Yes, exactly! Maybe the documentation is missing on this detail. Sorry for that. It was a bit internal but should be more documented.

borrepp, 2024-05-15:
Great¡¡ 

Thank you so much¡

closed 2025-05-02 by alejoe91