SpikeInterface/probeinterface #273: Issue when rotating 3D probes by specifying the axis as "xy"/"yz"/"zx"

DrTaDa, 2024-05-13:
It is written in the docstring of the method .rotate of the class Probe that the axis is to be specified as a string:
```
axis : "xy" | "yz" | "xz" | None, default: None
            Axis of rotation.
            It must be None for 2D probes
            It must be given for 3D probes
```

Howver, that leads to an error:
```
File "/home/tanguy.damart/virt_env/mozaik/lib/python3.8/site-packages/probeinterface/probe.py", line 1422, in _rotation_matrix_3d
    axis = axis / np.linalg.norm(axis)
  File "<__array_function__ internals>", line 180, in norm
  File "/home/tanguy.damart/virt_env/mozaik/lib/python3.8/site-packages/numpy/linalg/linalg.py", line 2511, in norm
    x = x.astype(float)
ValueError: could not convert string to float: 'xy'
```

Which hapens because the function _rotation_matrix_3d expects a list or array instead of a string: https://github.com/SpikeInterface/probeinterface/blob/86e5fa45fca02a82b42c954b088a09d9cd7bb010/src/probeinterface/probe.py#L1409

alejoe91, 2024-05-13:
Hi Tanguy!

Yeah indeed this is a bug. We should convert the string to the proper rotation axis (e.g., `xy` -> [0, 0, 1]). Alternatively we can ask the user to pass directly a 3D axis as input. What do you think?

DrTaDa, 2024-05-13:
Hey Alessio :))

I think having a function that would transform the string in a list as you suggested would be best.

closed 2026-09-15 by alejoe91