SpikeInterface/probeinterface #276: Probe and .bin file (from kilosort)

Lorena-J, 2024-05-29:
Hi, I have a question about how the probe channels are related to the .bin file.
I am logging with a 64 channel probe and openephys. At the end I have 64 .continuous files.

The probe is connected to an adapter and two intans (each intan with 32 channels). I don't quite understand how it recognises the geometry of the probe, since I understand that the .continuous files (called ch_X.continuous where X = channel number) will be indicating the channel of the intan, but how does it know which channel of the intan corresponds to the channel of the probe?

That is to say, if channel 1 (with coordinates 1,1) of my probe is connected to channel 3 of the intan, I understand that the .continuous file is ch_3.continuous. How kilosort knows that the ch_3.continuous is equivalent to the channel 1( coordinates 1,1) of my probe?

zm711, 2024-05-30:
Howdy @Lorena-J,

That's a wonderful question and actually the whole reason this library exists!

I recently made a notebook explaining this whole process [here](https://github.com/SpikeInterface/SpikeInterface-Training-Edinburgh-May24/tree/main/hands_on). You can actually run the examples on your computer and see if you understand each step. If not then you can post additional questions here. How does that sound?

alejoe91, 2025-05-02:
Closing for inactivity

closed 2025-05-02 by alejoe91