SpikeInterface/probeinterface #285: How to create a channel map for a microwire array electrode?

li1901, 2024-07-17:
My electrode is a 32-channel microwire array electrode, with the electrode site at the bottom end of each wire. I currently have two ideas: one is to use generate_multi_columns_probe. The other is to use generate_multi_shank, with only one site on each shank, but for generate_multi_shank I've only seen horizontally distributed multiple shanks and not a combination of horizontal and vertical distribution.
[image]

samuelgarcia, 2024-07-17:
Hi,
one way to create a probe from scratch is to use an excel, then read it with pandas and then create the probe using
`Probe.from_dataframe()`

Usin this approach is simpler for custum probes than using internal probe generator.

Zach made a cool tuto on this video around 18''
https://www.youtube.com/watch?v=pHze_8s4Qak&t=4605s

alejoe91, 2025-05-02:
Closing for inactivity. @li1901 feel free to reopen if this is not solved on your end