SpikeInterface/probeinterface #301: ASSY-77 > Adpt.A64-Om32x2 > 2x RHD2132 Pathway Wiring Discrepancy

bpqle, 2024-09-25:
Hi there,

My lab has been using the Cambridge probe ASSY-77 with the A64 adaptor connected to 2 RHD2132 chips and then to the OpenEphys acquisition board. I've been using the wiring map from probeinterface to do spike sorting on Kilosort.

Recently, Cambridge informed us that they shipped us a "mirrored" A64 adaptor initially, which led me to manually check the adaptor to RHD chip wiring and found some potential mistakes with the A64 adaptor -> 2 RHD2132 pathway wiring. I'll start with the normal (not mirrored) adaptor wiring first.

For reference, I'm using the [adaptor channel map](https://www.cambridgeneurotech.com/assets/files/ADPT-A64-Om32x2-map.pdf) provided by Cambridge's website and the [RHD2132 pin out map](https://intantech.com/RHD_headstages.html?tabSelect=RHD32ch&yPos=0) from intan's website. In the image below I've laid them out so that the RHD chip wiring can be overlaid directly on top of the adaptor wiring without flipping. The top image on the left shows what I assume is connected adaptor + chips arangement used for wiring (both chips facing inwards), as it provides the least amount of discrepancy.  (1) is how the RHD chip wiring should be arranged, with 0 starting in the middle on one side and wrapping around the other side incrementally before ending back. **When I traced the wiring pathway provided by probeinterface (2), there is a 4x4 block in the middle that is flipped (14..17, 30..1, 33..62, 46..49)** I hope this clearly demonstrate what I think is a discrepancy in the wiring pathway - please let me know if there is something I'm missing.

[image]

As an additional point, my lab actually uses the adaptor+chip arangement in (3), as otherwise the SPI cable connected to the chips will press against each other too much. Secondly, which chip ends up being 0..31 and the other 32..63 also depends on the order of the SPI cable plugged into the acquisition board (this causes a similar outcome to using a "mirrored" adaptor as we have been, I believe), so there are 2 further sources of confusion for the wiring. It would be nice if these could somehow be clarified when using the pathways to help new users - spikeinterface+probeinterface have already been very helpful for my research already, and I appreciate this project a lot.

samuelgarcia, 2024-09-26:
Hi.
Thanks for this feedback.

I guess you are mentioning this pathway

https://github.com/SpikeInterface/probeinterface/blob/main/src/probeinterface/wiring.py#L35

I remember that this pathway was quadriple checked inpedently by Alessio, Jessie (from Cambridge neurotech) Pierre Pascal (the end user) and me.
Maybe there are several version of this connector. I am not sure. 

In `'ASSY-77>Adpt.A64-Om32_2x-sm-NN>two_RHD2132'` the "NN" is for neuronexus. Maybe cambridge neurotec also have there adaptor which have a diferent wiring. Could you check this ?

bpqle, 2024-09-26:
Ah, I see that I've missed the NN part of the wiring pathway, that it's for the NeuroNexus adaptor rather than the Cambridge adaptor. 

When I'm looking at the NeuroNexus adaptor, though, the wiring pathway still doesn't make sense - actually less so than if a Cambridge adaptor were used instead. This is the adaptor pin out closest in name to the wiring pathway: Adpt-A64-OM32x2-sm that I can find on the NN website:

[image]

The quickest check is the left and right edge lines, as these will get connected to RHD lines differing by only 1. Take the bottom 2 lines on the left in the adaptor for example: 53 and 48. These, corresponding to the wiring in the wiring.py file, will be lines 53 and 15, respectively, on the acquisition pinout.

If I were using the Cambridge adaptor, the bottom 2 lines on the left will be lines 15 and 16 on the adaptor, which corresponds to lines 39 and 40 in the wiring pathway - this actually makes sense given how the RHD chip is laid out sequentially. Lines 15 and 16 on the NN adaptor are actually pretty far apart, so wouldn't get side-by-side line number on the RHD chip output. 

I think the fact that so many lines in the wiring pathway are correctly connected with the Cambridge adaptor is what threw me off to begin with.

zm711, 2024-10-08:
Sorry for the delay we just went through the map again together and they look good for using the neuronexus. We think your confusion is stemming from the fact that neuronexus adaptors have an extra step compared to the Cambridge neurotech adaptors. We plan on sharing a google sheet which goes over the mapping process soon for people to check.

That being said we are working on making maps for the RH2164 to cambridge adaptor as well as RH2132 adaptor (non-mirrored of course) now that we know these exist! Thanks for bringing it to our attention that these are necessary paths to add!

samuelgarcia, 2024-10-09:
Hi @bpqle

 here are the document we use to make the wiring:
[The schema](https://docs.google.com/presentation/d/18sMWuAIgtgmPcJw-0TC9f3JDNe64icD7pVA97QaMiuc/edit?usp=sharing)
[The wiring table](https://docs.google.com/spreadsheets/d/1Q6eAaRmn2c18Q_3H2A5WFX_ecxnq3gejkc2MP3pDMg4/edit?usp=sharing)

In the table please note that :
 * contact_id is the number on the probe (one based)
 * neuronexus_id is the number on the adaptor (one based)
 * headstage is which headstage index (0 or 1)
 * intan 0-31: number on intan per headstage (zero based)
 * channel_device_index final wiring (zero based)

This wiring is quite complex because it need a double 'jump'

zm711, 2024-10-09:
Just to help follow up on what Sam is saying. This double-jump is not required for Cambridge Neurotech probe to Cambridge Neurotech adaptor, so for mapping one must do two different strategies depending on the type of adaptor.

bpqle, 2024-10-10:
Thank you both for your responses. This cleared things up perfectly for me, as I hadn't realized there was a jump between probe and adaptor in the NN case. I also got to talk to Jessie from Cambridge at SfN recently to get a better understanding. 

I'll close the issue here. I wouldn't mind doing a PR for the Cambridge adaptor wiring map with the configuration I currently have. Again, thank you for the work you're doing.