SpikeInterface/probeinterface #311: Loading error for NP1016 probe version

meszena, 2024-12-20:
This has probably been done before, but it hasn't been solved for me yet, so I would like to ask you a short question: I am trying out a new human NP dataset recorded with the newest NP1016 version (same as NP1015 - basically a "2 column 1.0", just sterile, so it received a novel serial number), and I got this error below. I searched for it here and on the NP Slack channel and I saw other people had the same issue. You suggested installing python-neo and probeinterface. I did that but I am still getting this error. What could be a fix? Thanks in advance!
Domi
[image]

alejoe91, 2024-12-20:
H @meszena!

Can you make sure you have the latest probeinterface version? It seems that that probe is supported already: https://github.com/SpikeInterface/probeinterface/blob/0.2.24/src/probeinterface/io.py#L1090

meszena, 2024-12-20:
Thanks! I don't know what went wrong for the first time, I redid the install once more and now it's loading nicely, thanks for the quick reply, we can close this issue, I guess!

closed 2024-12-20 by alejoe91