SpikeInterface/probeinterface #321: The probeinterface.testing module loads probe.json.schema, which doesn’t ship with the wheel

musicinmybrain, 2025-02-06:
The new module `probeinterface.testing`, which ships as part of the `probeinterface` wheel, loads `probe.json.schema` at import time, but this only works using relative paths in a git checkout; `probe.json.schema` doesn’t ship in the wheel at all.

Separately, `probeinterface.testing` imports `jsonschema`, which isn’t a regular dependency, only a `test` extra dependency. I guess that’s “more OK:”

```
$ python3 -m venv _e
$ . _e/bin/activate
(_e) $ pip install probeinterface
$ python3 -c 'import probeinterface.testing'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
    import probeinterface.testing
  File "/home/ben/tmp/_e/lib64/python3.13/site-packages/probeinterface/testing.py", line 5, in <module>
    import jsonschema
ModuleNotFoundError: No module named 'jsonschema'
(_e) $ pip install probeinterface[test]
(_e) $ $ python3 -c 'import probeinterface.testing'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
    import probeinterface.testing
  File "/home/ben/tmp/_e/lib64/python3.13/site-packages/probeinterface/testing.py", line 8, in <module>
    schema = json.load(open(json_schema_file, "r"))
                       ~~~~^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/home/ben/tmp/_e/lib64/python3.13/resources/probe.json.schema'
```

I’m not quite sure what would make this “nicer” in the most straighforward way. If it’s feasible to move `probeinterface.testing` back out of `probeinterface` and into `tests/`, that would probably be best. Where it is, it looks like a public API, and that really doesn’t seem to be the intent.

alejoe91, 2025-02-07:
Hi @musicinmybrain 

I think we can ship the json.schema with the package and extend the `test` installation to add `jsonschema` without moving anything. How does that sound?

musicinmybrain, 2025-02-07:
> Hi [@musicinmybrain](https://github.com/musicinmybrain)
> 
> I think we can ship the json.schema with the package and extend the `test` installation to add `jsonschema` without moving anything. How does that sound?

Sure, that seems like a reasonable idea too. Thanks.

PR #326 (2025-03-07): Add json schema to core and make jsonschema dependency optional
Fixes #321 

@musicinmybrain can you test this solution? 

I moved the `probe.json.schema` to the package directory and added it to be included in the wheel.
In addition, the `jsonschema` package is only imported when calling `validate_probe_dict`, so it should be safer.

Let me know what you think!