SpikeInterface/probeinterface #324: Issues with Probe Handling in probeinterface v0.2.25

AmaliaLambertRinguet, 2025-03-05:
I am trying to use **SpikeInterface** for spike sorting with **Kilosort4**, but I am encountering several issues with **probe handling** and **setting the probe** on the recording object. Below is a summary of the steps and errors I have encountered.

- **Kilosort4 Setup**:
At first, I thought the problem was related to the sorting step in the 8-step pipeline provided by Samuel Garcia, specifically when calling sorting_KS4 = ss.run_sorter('kilosort4', recording). Here's the code I initially had for setting up the pipeline: 
`import spikeinterface.full as si
import spikeinterface.extractors as se
import spikeinterface.preprocessing as spre
import spikeinterface.sorters as ss
import spikeinterface.postprocessing as spost
import spikeinterface.qualitymetrics as sqm 
import spikeinterface.comparison as sc
import spikeinterface.exporters as sexp 
import spikeinterface.curation as scur 
import spikeinterface.widgets as sw
import matplotlib.pyplot as plt
import numpy as np

recording = se.BinaryRecordingExtractor(
    'Z:/Alexiaa/Electrophy cerveau/ChR2/dat/fichier_continu1722rec1.dat', 
    num_channels=32, 
    sampling_frequency=30000,  
    dtype=np.int16  
)

#plotting
si.plot_traces(recording, backend="ephyviewer")

#probe setup
import probeinterface as pi
manufacturer = 'cambridgeneurotech'
probe_name = 'ASSY-37-Fb'

probe = pi.get_probe(manufacturer, probe_name)
print(probe)

#preprocessing
recording = spre.bandpass_filter(recording, freq_min=300, freq_max=6000)
recording = spre.common_reference(recording, reference='global', operator='median')

#sorting
sorting_KS4 = ss.run_sorter('kilosort4', recording)  # Error occurs here`

I encountered the error "Exception: The sorter kilosort4 is not installed. Please install it with: pip install kilosort --upgrade" but, since I already had Kilosort4 installed and functional, I thought the issue seemed related to the environment access in Jupyter Notebook. 

- **Environment resolution**
I thus activated the Kilosort4 environment and reinstalled spikeinterface with : `conda activate kilosort4
pip install spikeinterface
jupyter notebook
` in Anaconda prompt. 

- **Probe handling issues**
After selecting the kernel Python(kilosort4) and re-running the code, a new error appeared: "Exception: There are no channel locations."
That's when I got that, despite being correctly identified by `pi.get_probe`, the issue was that the probe was not properly associated with the recording. 
I attempted different methods to set the probe with the recording (but none of them worked), using `recording.set_probe(probe)` # Error: 'module 'probeinterface' has no attribute 'read_probe' or by passing the probe file path before : `import probeinterface as pi
probe_path = 'C:/Users/T6-JulietteV/Desktop/Probe/ASSY-37-Fb.json' 
probe = pi.read_probe(probe_path)
print(probe)  # Error: AssertionError: must give Probe`

- **Working code for probe handling:**
After checking the available functions in probeinterface using `print(dir(pi))`, I found that the correct function to read the probe file is : `probe = pi.read_probeinterface(probe_path)
print(probe)
`
This cell now works without errors. 

- **Current issue:**
Despite this, I am still facing the following issue:
`recording.set_probe(probe)` 
This line raises the same error: AssertionError: must give Probe. Additionally, when printing the probe object, I see: "<probeinterface.probegroup.ProbeGroup object at 0x000001C3E9E17040>"
Should I extract the Probe from the ProbeGroup object or something like that? Or is the problem deeper? 

I have confirmed that the probe object is correctly generated, and the issue lies in linking the probe to the recording object. Without this link, I am unable to proceed with spike sorting, and the process is blocked at the probe association step.

I am a beginner in coding, and I’m trying my best to follow the documentation. However, I’m encountering continuous errors, and I fear I might keep going in circles. I’d appreciate any guidance to resolve the probe association issue! 
Thank you!

alejoe91, 2025-03-05:
@AmaliaLambertRinguet you're not setting the probe to the recording!

You need:
```
recording = recording.set_probe(probe)
# prior to bandpass
```

AmaliaLambertRinguet, 2025-03-05:
I'm actually now trying to set the probe to the recording, prior to bandpass, but I'm still encountering the same issue... 

Thanks for yout help, I'll try to "wire" the probe, then :)

AmaliaLambertRinguet, 2025-03-05:
[image]

[image]

AmaliaLambertRinguet, 2025-03-07:
[image]

[image]

[image]

The probe is being attached with 1 shank instead of the expected 3 (as indicated in the .json file). Also, I don't get why I have to manually enter the data for the probe `locations`, `shank_ids`...) despite being already present in the .json file. The probe data seems to not be correctly reflected from the .json file when I attempt to associate it with the recording. 

How can I correctly associate the probe from the .json file to the recording automatically?

alejoe91, 2025-03-08:
To have multiple groups, you have to use: `set_probegroup(probegroup, group_mode="by_shank")` (@samuelgarcia @zm711 this should be default imo when there is a single probe!)

You don't need to enter the locations and the shank ids! Those are automatically loaded!
If they seem unsorted, that's because you're missing a wiring step (you're wiring with np.arange, which doesn't consider the actual pathway between the contact and the recording device). See here: 
https://probeinterface.readthedocs.io/en/main/examples/ex_11_automatic_wiring.html#sphx-glr-examples-ex-11-automatic-wiring-py

zm711, 2025-03-08:
> To have multiple groups, you have to use: set_probegroup(probegroup, group_mode="by_shank") (@samuelgarcia @zm711 this should be default imo when there is a single probe!)

Yeah that makes sense to me and I think users will expect that too. Default should be to think about shanks and then if the user wants to ignore shanks they should have to change the setting. I think we are at a point where we should add a warning for a version or so to say defaults have changed right?

closed 2025-03-17 by AmaliaLambertRinguet