SpikeInterface/probeinterface #325: "read_openephys" encounters problems when load the setting.xml from the onebox device

jialeren2024, 2025-03-06:
```
import probeinterface as pi
Probe = pi.read_openephys('E:\open ephys\JM_m002\\2025-02-26_16-14-00\Record Node 101\settings.xml')
Probe
```
---------------------------------------------------------------------------
Exception                                 Traceback (most recent call last)
Cell In[25], [line 2](vscode-notebook-cell:?execution_count=25&line=2)
      [1](vscode-notebook-cell:?execution_count=25&line=1) import probeinterface as pi
----> [2](vscode-notebook-cell:?execution_count=25&line=2) Probe = pi.read_openephys('E:\open ephys\JM_m002\\2025-02-26_16-14-00\Record Node 101\settings.xml')
      [3](vscode-notebook-cell:?execution_count=25&line=3) Probe

File c:\Users\np\anaconda3\envs\si_env\Lib\site-packages\probeinterface\neuropixels_tools.py:970, in read_openephys(settings_file, stream_name, probe_name, serial_number, fix_x_position_for_oe_5, raise_error)
    [968](file:///C:/Users/np/anaconda3/envs/si_env/Lib/site-packages/probeinterface/neuropixels_tools.py:968) if neuropix_pxi is None:
    [969](file:///C:/Users/np/anaconda3/envs/si_env/Lib/site-packages/probeinterface/neuropixels_tools.py:969)     if raise_error:
--> [970](file:///C:/Users/np/anaconda3/envs/si_env/Lib/site-packages/probeinterface/neuropixels_tools.py:970)         raise Exception("Open Ephys can only be read when the Neuropix-PXI plugin is used")
    [971](file:///C:/Users/np/anaconda3/envs/si_env/Lib/site-packages/probeinterface/neuropixels_tools.py:971)     return None
    [973](file:///C:/Users/np/anaconda3/envs/si_env/Lib/site-packages/probeinterface/neuropixels_tools.py:973) if "NodeId" in neuropix_pxi.attrib:

Exception: Open Ephys can only be read when the Neuropix-PXI plugin is used

alejoe91, 2025-03-07:
@jialeren2024 is it ok if I include the settings file you provided so we can test against it?

jialeren2024, 2025-03-08:
Of course！

jialeren2024, 2025-03-08:
> [@jialeren2024](https://github.com/jialeren2024) is it ok if I include the settings file you provided so we can test against it?

Of course！Thanks for help！

alejoe91, 2025-03-10:
@jialeren2024 can you test this PR? #327

jmcdaniel2, 2025-03-11:
Hi I am also from the same lab that found this error and was looking to test this but am not familiar with how to get the PR, and I am striking out with Google. How would I be able to get the PR to test?

alejoe91, 2025-03-11:
I merged to main since it works with the test file that @jialeren2024 provided.

@jmcdaniel2 to install from source you should do the following:
```
pip uninstall probeinterface
pip install git+https://github.com/SpikeInterface/probeinterface.git
```

closed 2025-03-11 by alejoe91

jialeren2024, 2025-03-11:
> [@jialeren2024](https://github.com/jialeren2024) can you test this PR? [#327](https://github.com/SpikeInterface/probeinterface/pull/327)

It works! thank you!

PR #327 (2025-03-11): Support OneBox processor for Neuropixels-Open Ephys
Fixes #325