SpikeInterface/probeinterface #337: probe not ordered by 'device_channel_indices'

RobertoDF, 2025-04-16:
Hi,

I am loading a json file exported from ONIX using read_probeinterface. I see that the probe loaded is not ordered by  'device_channel_indices'. I suspect this is the reason for some downstream problems (waveform extraction fails). 

```
probes = read_probeinterface(json_file)

raw_rec = raw_rec.set_probegroup(probes)
```

how can I order the probes object by  'device_channel_indices' before assigning it?

Thanks!

RobertoDF, 2025-04-16:
the json exported from onix  is ordered by "contac_ids" and not  'device_channel_indices' . 

[image]

alejoe91, 2025-04-16:
Hi @RobertoDF 

You don't need to reorder the probe. When setting a probe/probegroup, SpikeInterface reorders by device_channel_indices automatically ;)

alejoe91, 2025-04-16:
You can check with:
```
pg = raw_rec.get_probegroup()
pg.to_dataframe()
```

RobertoDF, 2025-04-16:
is that possible to do easily?

alejoe91, 2025-04-16:
Are you getting any errors? If not then don't worry about it :)

RobertoDF, 2025-04-16:
The main problem is that when  I extract the waveforms, thay are all flat. 

[image]

RobertoDF, 2025-04-16:
no, I just do 
```
 raw_rec = read_binary(path_recording,
                              sampling_frequency=fs_hz,
                              dtype=np.uint16,
                              num_channels=num_channels,
                              gain_to_uV=gain_to_uV,
                              offset_to_uV=offset_to_uV)

        probes = read_probeinterface(probe_json)

        raw_rec = raw_rec.set_probegroup(probes)
        raw_rec = spre.scale_to_uV(recording=raw_rec)
```

RobertoDF, 2025-04-16:
I just noticed that they are not totally flat, this is the biggest cluster by amplitude 

[image]

they are just unusually small.

RobertoDF, 2025-04-17:
am I correct in thinking that it is enough to high pass at the line of the sorting analyzer?
`create_sorting_analyzer(sorting, spre.highpass_filter(recording=sub_rec)s)`
kilosort is already doing it by itself when sorting.

closed 2025-04-17 by RobertoDF