SpikeInterface/probeinterface #342: hatchling build broke pypi wheel

alejoe91, 2025-05-05:
@h-mayorquin 

I think that https://github.com/SpikeInterface/probeinterface/pull/335 broke pypi packaging. After installing the latest just released version I get a `probeinterface` not found.

And in fact, the shipped whl doesn't contain anything meaningful. I'll try to find a solution

closed 2025-05-05 by alejoe91

h-mayorquin, 2025-05-05:
Thanks for fixing this. That's strange that this was not captured by the the full tests which install the latest version. Maybe this is related to the fact that we test editable and that's why the error was not catched?

h-mayorquin, 2025-05-14:
Here, maybe this will help to detect other errors in the future anyway:

https://github.com/SpikeInterface/probeinterface/pull/348

PR #343 (2025-05-05): Fix hatchling packaging
Fixes #342